git-artes/gr-isdbt #2: 1-seg output

rfrht, 2015-08-24:
Hello,

Do you have any tests using 1-seg output and RTL2832U dongles?

Thanks for the project!!

git-artes, 2015-08-24:
Hello, 
Do you mean using a dongle to receive the 1-seg segment only? Then, sadly the answer is no (yet). We are currently working in implementing a better OFDM receiver (we will release it shortly), and we will probably try to tackle that next. If you have any ideas or implementations they are more than welcome!
Best
Federico

rfrht, 2015-08-24:
Hola Federico!

Thank you very much for your (super-quick) reply.

RTL2832U SDR dongles (please visit [Osmocom](http://sdr.osmocom.org/trac/wiki/rtl-sdr)) are very cheap and common in hacker scene. However, its limited bandwidth (~ 3 MHz) makes it unsuitable for a full-seg ISDB-T (6 MHz, as per [Wikipedia](https://en.wikipedia.org/wiki/ISDB-T_International#Technical_facts)), making it unsuitable for this hardware.

However a [1-Seg](https://en.wikipedia.org/wiki/1seg) decoding would be suitable for such hardware.

Maybe we could set this issue as a RFE (Request For Enhancement)? :-)

Cordiales saludos,

- Rodrigo.

git-artes, 2015-08-24:
Hi Rodrigo, 
We got several dongles (those sold at NooElec BTW) a couple of months ago, and what you describe was precisely one of our ideas. However, with the beginning of classes here our time for the project has somewhat decreased. 
I've labelled this issue as "enhancement", and we will tackle it ASAP, but do not expect anything less than some (few I hope) months. 
best
Federico

git-artes, 2015-12-04:
Hi Rodrigo, 
As promised, and some 4 months later, the feature is now implemented. I've tested it just now, and it worked with my NooElec dongle. Please see the new rx_1seg_demo.grc example. 
Let us know if it worked for you. 
Best
Federico

rfrht, 2015-12-06:
Hello, Federico!

First of all, I would like to **congratulate and thank** you and your team! Thanks for putting time and passion to make it happen! Hope that seed further developments!

It worked like a charm with Gnuradio [Live DVD](http://gnuradio.org/redmine/projects/gnuradio/wiki/GNURadioLiveDVD) version 3.7.8.1. Zero compilation errors, warnings or whatsoever. Crystal-clear compilation.

Now, I'm struggling to find the right frequency for the 13th segment for the 1-seg output.

This is the list of central frequencies broadcast here in Sao Paulo: https://pt.wikipedia.org/wiki/Lista_de_canais_da_televis%C3%A3o_digital_brasileira#S.C3.A3o_Paulo_.5B2.5D

Wondering if do you have a calculation to find out the correct frequency for a given channel?

Thank you very much!

- Rodrigo.

git-artes, 2015-12-07:
Hi Rodrigo, 
Glad it compiled, but I would be "gladder" if it'd work. I'm guessing not yet. 
Regarding your question about the 13th segment, the 1-seg transmission is always on the central segment (segment 0). I have not read the norm to check if it is compulsory, but all figures show it in the central segment. So you should tune your receiver to the central frequency. 
Let me know if you got it running. 
BTW, I've checked your pull request. Thank you very much for the contributions. I will answer it in the comment section. 
best
Federico

closed 2016-02-26 by git-artes