git-artes/gr-isdbt #3: fullseg_receiver_and_measurements missing MER blocks

joharivillegas, 2015-10-29:
Hi Good day! 
I have downloaded the your files and tried to open the grc files in there. However i came across a problem when i opened the fullseg grc file. it has a missing block. could you please enlighten me about this. Thank you. 
[image]

git-artes, 2015-10-29:
Hello Johari, 
Thanks for the interest in the project. Those blocks come along our accompanying OOT module gr-mer. It may be downloaded and installed from https://github.com/git-artes/gr-mer. I will update the README to comment on this. 
Please let us know if the receiver worked for you. This kind of feedback is more than welcome. 
best
Federico

joharivillegas, 2015-10-29:
Thank you so much sir Federico. We will try to run it back.

joharivillegas, 2015-11-06:
Good day sir Federico. I tried to run it back and its output is shown below. 
[image]

I'm having a hard time interpreting these data. would you shed a light to me sir if what does these means and how can we properly have a good reception? Thank you so much.

git-artes, 2015-11-06:
Hi Johari, 
What you are obtaining means that your input does not have an ISDB-T signal or (more probably) you have configured transmission parameters incorrectly (cyclic prefix length and/or mode). 
What input are you using? A live one from the USRP or one of our recordings from http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/? 
best
Federico
PS since the problem with MER was resolved, I am closing this issue. Please open a new one when you have new problems, or to keep me up to date with this one in particular.

closed 2015-11-06 by git-artes