git-artes/gr-isdbt #4: fullseg_receiver_and_measurements input live from usrp

joharivillegas, 2015-11-07:
Good day sir federico. in relevance with the last image i sent to you,
what we were doing was we receive the isdbt-signal from a local tv broadcast station here in philippines via the usrp1 with wbx 50-2200 mhz daughterboard is gnu radio. I am also confused if whether by what you mean using your recording as our input, would that mean that we will be needing an isdbt transmitter as well?  Please enlighten us. 
Thank you so much. 
Regards, 
Johari

git-artes, 2015-11-07:
Hello Johari, 
Regarding the image you sent me, then you are probably incorrect in either the cyclic prefix or the mode. You should use our example "obtaining_the_parameters.grc" to obtain them. Please refer to the README. 
Regarding my suggestion of using our recording as an input, what we are sharing in http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/ are the raw complex baseband samples we have obtained by recording some local transmissions here in Uruguay (we have used them extensively and they work perfectly with our receiver). They may be used as an input in all the examples via a "File Source" block. This possibility is included in all the examples (although I believe the coresponding blocks are disabled). 
best
Federico

joharivillegas, 2015-11-12:
Hi sir Federico! Good day
Regarding with your request of a picture of the spectrum. 
[image]
The Isdb-t signal of ABS-CBN(local tv operator) is centered at 611.143MHz. we have obtained the spectrum by using the obtaining the parameters grc file. 

Regarding with the correct configuration of the parameters, we are yet to go to ABS-CBN to allow as assess in obtaining the parameters. since he are having some confusions with the grc file obtaining the parameters if what we obtained is correct or not. 

Regards, 
Johari

git-artes, 2015-11-13:
Hi Johari, 
Using the interface of obtaining_parameters.grc, you should try different combinations of Guard Interval (1/4, 1/8, 1/16, 1/32) and Mode (1, 2 or 3). If there is a an OFDM signal (with the correct sampling, and without too many dropped samples) a "peak" should be obtained on the first graph, as the one I show here. 
[image]

If you obtain noise, as in the graph you are showing, then you either have configured an incorrect mode, or you don't have an OFDM signal. I believe, by the signal levels you are obtaining, that the latter is the case. Could you please show us a picture of what you obtain by sampling the spectrum at 8MS/s centered around that frequency? (you may use uhd_fft for instance). 
best
Federico

joharivillegas, 2015-12-17:
Hi sir Federico, Good day!
 
so far in our experimentation, we varied different guard intervals and mode for us to obtained a "peak" of the received isdbt signal, however we were not yet able to obtain it. i am confused by what you mean uhd_fft, on how are we going to use it. 
However, we have determine the parameter of the local tv station that we were trying to received
attached here. 
would there be a difference if we will be using usrp1 rather that usrp n200? we have both here in our lab.
[image]

Thank you Sir Federico,

Regards,
Johari

git-artes, 2015-12-17:
Johari, 
uhd_fft is the simple spectrum analyzer included in the installation of UHD and GNU Radio. 
What you attached is mostly pertaining to the MPEG stream. In our case the only important parameters are those regarding transmission (mode, CP length, etc.). 
I'm not sure about changing hardware. 
best

arvintateyu, 2016-01-28:
Good day sir Federico!

I am Arvin, a co-proponent of Johari for the study we are currently working on. I have read the comments and I have obtained this signal in UHD_FFT.grc. 
[image]

I would like to ask why i couldn't obtain this signal in obtaining parameters.grc? Does it have something to do with the OFDM SYNCHRO peak over/under average? I apologize for the inconvenience I caused by sending repeated messages in gmail and in here :(

git-artes, 2016-01-29:
Johari and Arvin, 
I am currently on vacation. I'm assuming you've got enough time to read the code and finally make it work with our recordings. 
Regarding your signals, they do not look at all like OFDM. There is something there, but is definitively not ISDB-T (it should have a 6-MHz bandwidth and be mostly constant). You should check the transmission frequency of your local TV channels. 
best

closed 2016-02-26 by git-artes