git-artes/gr-isdbt #11: Unable to run 1-seg RX demo

swineone, 2016-03-18:
I'm trying to run the 1-seg RX demo of gr-isdbt (examples/rx_1seg_demo.grc) on a MacBook Pro running Mac OS X 10.11.3, as well as the latest GNU Radio Live DVD (based on Ubuntu 14.4.3). The GNU Radio version is 3.7.9.1 in both cases. I've compiled gr-isdbt from source using the latest commit at the time of writing (2d209c4). I tried using both an RTL-SDR dongle and the 569MHz_recording.tar.gz file that can be downloaded from http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/recordings/569MHz_recording.tar.gz. In both cases, some kind of processing is going on, the constellation display is updated and so on, but the test_out.ts file stays at 0 bytes forever. Please advise.

git-artes, 2016-03-18:
Hi,
You're probably misconfiguring some of the upstream blocks. The reed solomon block does not output anything in that case. Those parameters may be checked with the tmcc decoder (configure it so that it prints the parameters).
Best
Federico

swineone, 2016-03-18:
I'm very new to GNU Radio so this may be indeed the case. All I did was open the rx_1seg_demo.grc file, double click on the "File Source" block, point it to the place where I downloaded the recording, and change "Repeat" to No.

I tried enabling "Print Params" in the TMCC decoder, but it seemed to make no difference to what is printed in the console output. For reference, here are the first few lines:

> Generating: '/Users/decio/Downloads/gr-isdbt/examples/rx_demo.py'
> 
> Executing: /usr/local/opt/python/bin/python2.7 -u /Users/decio/Downloads/gr-isdbt/examples/rx_demo.py
> 
> Using Volk machine: avx_64_mmx
> Segments: 1
> d_noutput: 384
> gr::buffer::allocate_buffer: warning: tried to allocate
>    348 items of size 188. Due to alignment requirements
>    1024 were allocated.  If this isn't OK, consider padding
>    your structure to a power-of-two bytes.
>    On this platform, our allocation granularity is 4096 bytes.
> gr::buffer::allocate_buffer: warning: tried to allocate
>    321 items of size 204. Due to alignment requirements
>    1024 were allocated.  If this isn't OK, consider padding
>    your structure to a power-of-two bytes.
>    On this platform, our allocation granularity is 4096 bytes.
> gr::buffer::allocate_buffer: warning: tried to allocate
>    321 items of size 204. Due to alignment requirements
>    1024 were allocated.  If this isn't OK, consider padding
>    your structure to a power-of-two bytes.
>    On this platform, our allocation granularity is 4096 bytes.
> previous_symbol: 0, 
>  current_symbol: 2
> d_corr_sp = {131.254227, 217.050125, 317.050568, 52.914509}
> OFDM_SYNCHRO: peak under/over average! peak -1.136172, avg_max -0.674585, avg_min -0.777794
> previous_symbol: 2, 
>  current_symbol: 1
> d_corr_sp = {75.283516, 334.854248, 156.149429, 63.200245}
> OFDM_SYNCHRO: peak under/over average! peak -0.929594, avg_max -0.641050, avg_min -1.084656
> OFDM_SYNCHRO: peak at border! peak -0.929594, avg_max -0.641050, avg_min -1.084656, peak_index: 15
> OFDM_SYNCHRO: peak under/over average! peak -0.780521, avg_max -0.641050, avg_min -0.980606
> VITERBI: d_init=True
> OFDM_SYNCHRO: peak under/over average! peak -0.778441, avg_max -0.641050, avg_min -0.861125
> OFDM_SYNCHRO: peak under/over average! peak -0.755186, avg_max -0.641050, avg_min -0.934647
> OFDM_SYNCHRO: peak under/over average! peak -0.755186, avg_max -0.641050, avg_min -0.972787
> previous_symbol: 1, 
>  current_symbol: 3
> d_corr_sp = {253.492355, 163.112000, 264.092224, 405.427734}
> OFDM_SYNCHRO: peak under/over average! peak -1.054460, avg_max -0.729244, avg_min -1.014069
> previous_symbol: 3, 
>  current_symbol: 2
> d_corr_sp = {263.085663, 82.861786, 609.761292, 151.675232}
> OFDM_SYNCHRO: peak under/over average! peak -1.252648, avg_max -0.571403, avg_min -0.858780
> OFDM_SYNCHRO: peak under/over average! peak -0.767336, avg_max -0.571403, avg_min -1.022610
> OFDM_SYNCHRO: peak under/over average! peak -0.735650, avg_max -0.571403, avg_min -1.089492
> OFDM_SYNCHRO: peak under/over average! peak -0.735650, avg_max -0.571403, avg_min -1.041195
> OFDM_SYNCHRO: peak under/over average! peak -0.749995, avg_max -0.571403, avg_min -1.145790
> OFDM_SYNCHRO: peak under/over average! peak -0.749995, avg_max -0.571403, avg_min -1.100455
> OFDM_SYNCHRO: peak at border! peak -0.749995, avg_max -0.571403, avg_min -1.100455, peak_index: 2
> previous_symbol: 2, 
>  current_symbol: 0
> d_corr_sp = {218.159836, 112.232018, 55.699749, 207.933304}
> OFDM_SYNCHRO: peak under/over average! peak -0.994152, avg_max -0.670031, avg_min -1.268404
> previous_symbol: 0, 
>  current_symbol: 3
> d_corr_sp = {325.498779, 377.224060, 392.211456, 574.101074}
> OFDM_SYNCHRO: peak under/over average! peak -0.993341, avg_max -0.591788, avg_min -1.338741
> previous_symbol: 3, 
>  current_symbol: 3
> d_corr_sp = {57.520016, 154.947952, 275.355591, 310.860474}
> previous_symbol: 3, 
>  current_symbol: 3
> d_corr_sp = {271.633362, 258.840820, 212.560867, 369.453705}

From then on the output is very similar to these last few lines.

Any idea what I may be doing wrong?

git-artes, 2016-03-18:
If you want to run the 1seg example, you should use the dongle recording (the third one on the list on http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/). There you will some documents which may help you further.
Best

swineone, 2016-03-18:
Thanks for the pointer. The dongle recording worked OK, proving my GNU Radio and gr-isdbt installation is fine. I tried some more with the RTL SDR dongle, and I managed to receive two TV stations. It turns out that the station I was trying to receive, at 641.143 MHz (which is the one physically closest to me) is for some reason not working, but I assume this is a dongle issue of some kind. Either way, thanks for helping out.

swineone, 2016-03-20:
Just to report back that I managed to receive the station at 641.143 MHz by changing the guard interval to 1/8. I wonder if there's an automatic way of figuring out the correct guard interval, or should it be done by trial and error?

git-artes, 2016-03-20:
Hi, 
Glad it worked. Automatically obtaining those parameters is not possible at the current stage of gr-isdbt. However, it is very simple to obtain them and does not require a complete receiving chain. See obtaining_the_paramers.grc for such a flowgraph (in our website there is a sort of manual, in spanish, on how to use it). It works for full_seg, although it should be easily adaptable to the 1-seg case. 
best
Federico

swineone, 2016-03-20:
Thank you, and I'll close this again.

closed 2016-03-20 by swineone