git-artes/gr-isdbt #13: Problem in the make process

lpachecs, 2016-05-23:
Hello there,

I'm having a problem on the make...

Here is the terminal msg, I don't know what's wrong... could you please help me? Thanks.

[  2%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o
In file included from /usr/include/gnuradio/block.h:29:0,
                 from /root/gr-isdbt/include/isdbt/ofdm_sym_acquisition.h:35,
                 from /root/gr-isdbt/lib/ofdm_sym_acquisition_impl.h:34,
                 from /root/gr-isdbt/lib/ofdm_sym_acquisition_impl.cc:36:
/usr/include/gnuradio/logger.h:122:31: fatal error: log4cpp/Category.hh: No such file or directory
 #include <log4cpp/Category.hh>
                               ^
compilation terminated.
lib/CMakeFiles/gnuradio-isdbt.dir/build.make:54: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o] Error 1
CMakeFiles/Makefile2:123: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

git-artes, 2016-05-23:
Hi, 
It seems you don't have log4cpp installed. This is required by GNU Radio, and should have been installed along it. How have you installed GNU Radio? From source, PyBombs, or pre-compiled binaries (i.e. by using apt-get install)?
best

lpachecs, 2016-05-23:
Hi,

I've used the apt-get install gnuradio.

I'm thinking in installing this: http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGRFromSource#Using-the-build-gnuradio-script

Is it good?

git-artes, 2016-05-23:
Hi, 
As mentioned in the Remarks section of our README, you should also install gnuradio-dev. 
best

lpachecs, 2016-05-23:
But it is already installed...

git-artes, 2016-05-23:
Then, you probably got this bug: https://bugs.launchpad.net/ubuntu/+source/gnuradio/+bug/1358173. 
You should install liblog4cpp5-dev manually. 
best

lpachecs, 2016-05-23:
I've managed to install the package after installing log4cpp! XD

But now I'm having some missing blocks 

[image]

git-artes, 2016-05-23:
did you sudo make install?

lpachecs, 2016-05-23:
Yes, but I'm having a problem with sudo, so I installed as root. Is that ok?

git-artes, 2016-05-23:
so, you've become root via the "su" command and then performed "make install"? It should have worked. Did you also run "ldconfig"?

lpachecs, 2016-05-23:
Yes...
Now, everything is installing without any problem, but the blocks are still missing...

I noticed this errors when opening the examples:

Loading: "/home/pacheco/gr-isdbt/examples/rx_1seg_demo.grc"
>>> Error: Block key "isdbt_bit_deinterleaver" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "isdbt_byte_deinterleaver" not found in Platform - grc(GNU Radio Companion)

I receive this error for all missing blocks... Have you ever saw this?

git-artes, 2016-05-23:
what happens when you run "import isdbt" from a python terminal?

lpachecs, 2016-05-23:
Nothing happened...

I've already generated the code in examples/sym_align_ofdm.grc also.

lpachecs, 2016-05-23:
I think I got what's happening...

the Cmake is not working properly... It's reporting a bunch of errors...

I cmake as a SU and via sudo, but nothing worked...

git-artes, 2016-05-23:
lpachecs, I don't mean to be rude, but if you want our help, you'll have to be more explicit in your answers and explanations. What does "nothing happened" mean exactly? No errors in the output and the import was correctly performed? 
What are these "bunch of errors" that cmake is reporting?

lpachecs, 2016-05-31:
No problem man...
 After the cmake, I get the following msgs for all parameters:

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target
  "/home/pacheco/gr-isdbt/python/qa_tmcc_decoder_1seg.py".
Call Stack (most recent call first):
  python/CMakeLists.txt:64 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

After that, I run the sudo make and sudo make test, with the following results:
89% tests passed, 2 tests failed out of 19

Total Test time (real) =  19.31 sec

The following tests FAILED:
	  2 - qa_ofdm_sym_acquisition (Failed)
	 17 - qa_ofdm_synchronization (Failed)
Errors while running CTest
Makefile:117: recipe for target 'test' failed
make: *** [test] Error 8

And when I sudo make install, I get the following msg for all components:
Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/classgr_1_1isdbt_1_1ofdm__sym__acquisition__impl.js
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/classgr_1_1isdbt_1_1frequency__deinterleaver__1seg__impl-members.html
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/classgr_1_1isdbt_1_1reed__solomon__dec__isdbt-members.html
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/time__deinterleaver_8h.html
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/frequency__deinterleaver_8h__dep__incl.md5
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/classgr_1_1isdbt_1_1frequency__deinterleaver__1seg__inherit__graph.png
-- Up-to-date: /usr/include/gnuradio/share/doc/gr-isdbt/html/classgr_1_1isdbt_1_1energy__descrambler__impl__inherit__graph.map

Thanks for the help you are giving me.

git-artes, 2016-05-31:
Some tests don't pass, and I was about to correct that. Ignore them for now. Your next step should be to perform "sudo make install" and you should be all set to go. 
best

andregcoimbra, 2016-06-11:
HI,
I use VirtualBox with the below Linux and CPU. The gnuradio work normally. But when I try the make command occured the error below. I try all command as root and there i'snot dependecy

-------------

Linux kali 4.3.0-kali1-686-pae #1 SMP Debian 4.3.3-7kali2 (2016-01-27) i686 GNU/Linux

Architecture:          i686
CPU op-mode(s):        32-bit
Byte Order:            Little Endian
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht nx rdtscp constant_tsc xtopology nonstop_tsc pni ssse3 cx16 sse4_1 sse4_2 popcnt hypervisor lahf_lm

--------------

_root/gr-isdbt/lib/d_viterbi.c: In function ‘d_viterbi_chunks_init_sse2’:
/root/gr-isdbt/lib/d_viterbi.c:274:12: warning: SSE vector return without SSE enabled changes the ABI [-Wpsabi]
     mm0[i] = _mm_setzero_si128();
            ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:1249:0,
                 from /root/gr-isdbt/lib/d_viterbi.h:29,
                 from /root/gr-isdbt/lib/d_viterbi.c:32:
/usr/lib/gcc/i686-linux-gnu/5/include/emmintrin.h:753:1: error: inlining failed in call to always_inline ‘_mm_setzero_si128’: target specific option mismatch
 _mm_setzero_si128 (void)
 ^
/root/gr-isdbt/lib/d_viterbi.c:275:14: error: called from here
     pp0[i] = _mm_setzero_si128();
              ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:1249:0,
                 from /root/gr-isdbt/lib/d_viterbi.h:29,
                 from /root/gr-isdbt/lib/d_viterbi.c:32:
/usr/lib/gcc/i686-linux-gnu/5/include/emmintrin.h:753:1: error: inlining failed in call to always_inline ‘_mm_setzero_si128’: target specific option mismatch
 _mm_setzero_si128 (void)
 ^
/root/gr-isdbt/lib/d_viterbi.c:274:14: error: called from here
     mm0[i] = _mm_setzero_si128();
              ^
lib/CMakeFiles/gnuradio-isdbt.dir/build.make:254: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/d_viterbi.c.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/d_viterbi.c.o] Error 1
CMakeFiles/Makefile2:137: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2_

Best regards.

braselectron, 2016-06-13:
Dear Sir,

I also had the same problem when I installed your build for testing following the instructions.

My system is Ubuntu 14.04 32bit version
3.13.0-88-generic #135-Ubuntu SMP Wed Jun 8 21:10:37 UTC 2016 i686 i686 i686 GNU/Linux

Can you please explain how to uninstall the make so I can try again and clean it from the error make process ?   I need this information because I tried: 

`make uninstall`

at the build directory and I get this answer:

john:$~/gr-isdbt/build$ make uninstall
CMake Error at cmake_uninstall.cmake:4 (MESSAGE):
  Cannot find install manifest:
  "/home/john/gr-isdbt/build/install_manifest.txt"

I am researching on 1-seg reception using gnuradio and RTL2838U dongle

My installation of gnuradio-sdr is following this guide:

[http://gqrx.dk/download/install-ubuntu](http://gqrx.dk/download/install-ubuntu)

Thank you for your help.

John

git-artes, 2016-06-13:
@andregcoimbra, you should start a new issue when reporting a new problem. I've never stumbled upon such error, but it's true that the viterbi decoder uses machine instructions, so it's not guaranteed to work on all machines and setups. Maybe you could try what @braselectron suggests. 

Regarding your question @braselectron, I'd suggest you try rx_1seg_demo.grc and see if it works. 

In any case, I'm closing this issue. You are welcome to start a new one if you have further problems (or this one persists). 
best

lpachecs, 2016-06-14:
Hi there,

I finally was able to "activate" the blocks.

I had to put a line with the path for the local blocks in:

~/.gnuradio/config.conf

Added this line:

[grc]
local_blocks_path=/usr/local/share/gnuradio/grc/blocks

Saved the file and after that, the ISDBT blocks and functions were live.

Thanks for the support.

git-artes, 2016-06-14:
Glad it worked. Regarding the solution, such configuration should be included in /usr/local/etc/gnuradio/conf.d/grc.conf for a system-wide installation. Just out of curiosity, do you have such file? What does it say?
best

lpachecs, 2016-06-14:
I don't have this file here, the directory (/usr/local/etc) is empty.

git-artes, 2016-06-15:
If your installation was not system-wide, then you may have the file elsewhere. For instance, if GNU Radio was installed with pybombs, you may find it at ~/target/etc/gnuradio/conf.d/grc.conf

closed 2016-06-22 by git-artes