git-artes/gr-isdbt #14: DTV recording and TMCC displaying

eaiglesiasr, 2016-05-30:
Hi! I'd like to comment the following: I've been trying to record a TV RF signal from DTV local channel, using a USRP N200 and daughter TVRX2 card, in order to obtain its TMCC parameters through   "obtaining_parameters.grc" , but any successful parameter was received ( I attached three images)..so that I have the following questions: is it ok the way of recording of RF DTV signal o there's another way? is there any problem or special issue  in the received spectrum?  I will be  thanked if someone help me...

[image]
   
[image]
[image]

git-artes, 2016-05-31:
Hi, 
Thanks for the interest in the project. 
Now, regarding your questions. First, the examples we provide were tested with a USRP B100, whose sampling rate could be set at 8MS/s. You should make sure that this is the case with the N200 (typically, right at the beginning it will tell you in the standard output that the requested sampling rate was achieved or not). 
Second, spectrum looks good to me. 
Third, if the sampling rate is correctly set, and an ISDB-T signal is there, you should try to make a "peak" appear in the upper graph of the last picture you sent, by trying different values of guard interval and mode. If you understand spanish, you may read http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/papers/estudio_propagacion.pdf (in particular take a look at Figure 4). 
Please let me know if you got it working. 
best

eaiglesiasr, 2016-05-31:
Hi!
To start, I comment the following:
I've just  obtained the "peak" (I attached a image). I had a problem with sampling rate since USRP N200 has 100MS/s  of sampling rate and USRP B100 has 64MS/s., but when I probed "viewing_the_constellation" two cases happened when Interpolate parameter of ofdm_synchronization block was modified ( I attached two images). 
[image]
INTERPOLATE=YES 
[image]
INTERPOLATE=NO
[image]

git-artes, 2016-05-31:
That's weird. The difference between Interpolate yes and no, is whether to estimate and compensate for the sampling error. In any case, it should perform better. What sampling rate are you using now? Which parameters on the rational resampler?
best

eaiglesiasr, 2016-06-02:
Yes, I checked some connections and found a problem with antenna, now I can receive well Interpolate yes and no (sampling rate and parameters can be checked in the attached image).
[image]

 However, there's a problem  some seconds after running the file (two images are attached). 
 Could be published  your comments?
[image]
[image]

git-artes, 2016-06-02:
Hi, 
Glad it worked. Regarding the last problem, you are getting some "O"s, meaning your machine is not fast enough to process the whole flowgraph. Try enabling the Stream to Vector and Vector to Stream blocks and see if it works. 
best

eaiglesiasr, 2016-06-03:
Hi,
I tried enabling the Stream to Vector and Vector to Stream blocks, using "Num Items" matches 2M,  and saw that continues the problem (I attached two pictures), is "Num Items" matches 2M  accuracy?  
[image]
[image]
Furthermore, I tried to check sampling error, but a message was displayed...
[image]
Is it OK the way to measure o there's any other way?

git-artes, 2016-06-03:
Hi, 

The "stream to vector" -> "vector to stream" trick acts as a buffer, so eventually, if your machine is not fast enough, you will get overflows (as the "D"s indicate in the output). However, you have already obtained the parameters, and you should try using the rx_demo to check whether or not the system is able to decode the signal (obtaining_the_parameters.grc is very CPU intensive and not optimized at all). If not, you can always make a recording and see if it works. 

Now, regarding the last picture, you should read the error it reports: "Missing connection from output port 1". You should connect a null sink or something on the rest of the output ports to make it work. 

best
Federico

git-artes, 2016-06-13:
I'm closing the issue since no further feedback was received.

closed 2016-06-13 by git-artes