git-artes/gr-isdbt #15: rx_1seg_demo using RTL-SDR RTL2838U/R820T - AttributeError

braselectron, 2016-06-15:
I am trying to test the gr-isdbt using NooElec nano2+ but it stops with the following output (traceback):

Generating: '/home/john/gr-isdbt/examples/rx_demo.py'

Executing: '/usr/bin/python -u /home/john/gr-isdbt/examples/rx_demo.py'

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.009.004-release

**Using Volk machine: ssse3_32_orc**
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 

**Using device #0 Realtek RTL2838UHIDIR SN: 00000001**
Found Rafael Micro R820T tuner
Exact sample rate is: 1015873.002204 Hz

**Traceback (most recent call last):**
  File "/home/john/gr-isdbt/examples/rx_demo.py", line 341, in <module>
    main()
  File "/home/john/gr-isdbt/examples/rx_demo.py", line 329, in main
    tb = top_block_cls()
  File "/home/john/gr-isdbt/examples/rx_demo.py", line 226, in __init__
    self.isdbt_viterbi_decoder_0 = isdbt.viterbi_decoder(4, 1)
AttributeError: 'module' object has no attribute 'viterbi_decoder'

>>> Done (return code 1)

git-artes, 2016-06-15:
Hi, 
Have you updated your libraries (i.e. sudo ldconfig)?
best

braselectron, 2016-06-15:
Dear git-artes,

First of all, thank you.  Second, I did that before doing this post.  I repeated the `sudo ldconfig` two times more and it finally worked, I believe it did.   Hurrah!!!

I see this now:
[image]

Now is there a way to see a image frame also using your build ?

git-artes, 2016-06-15:
Glad it worked. Now, regarding how to see the actual multimedia signal, you should first check which parameters are being used by the transmitter. You may print them by using the Print params. option in the TMCC Decoder block. When you get those parameters, you should configure the rest of the blocks accordingly (notably the symbol demapper, viterbi decoder and time deinterleaver). 
The graph you get when running the flowgraph is the BER obtained by the Viterbi and RS blocks, so if everything is correctly configured, it should go to something like -3 or less (it is in logarithmic scale). 
So, when this is all set, you should get a Transport Stream file where the File Sink is pointing. You should be able to read it with ffplay or mplayer. You may even define it previously as a fifo (with the command mkfifo) and see it online without generating a big file. 
best
Federico

braselectron, 2016-06-15:
Actually I did use that file to play the reception stream with VLC player and it did not work, so I used the avplay and it worked.  Please see screenshot:

[image]

I did also create a `mkfifo mpeg.ts` to avoid the impact of writing the output to disk but it did not work.
This is what I did: made the fifo link and changed the File Sink block and tried  avplay and VLC decoding but didn't work.

I than tried to use de UDP Sink block to stream to VLC and also did not work. 

Also no sound.

git-artes, 2016-06-15:
Hi, 
I've never been able to make VLC work with the Transport Streams, I don't know why. What has worked best for us is ffplay, included as part of ffmpeg. Second best is mplayer. I've never tried avplay. 
Regarding why it works with a file, and not with a FIFO, sometimes I have to wait a long time before the video is played. Or sometimes, I execute the flowgraph with a FIFO as its File Sink, play the file with ffplay, wait a while and close ONLY the flowgraph, and then the video is shown. 
This weird behavior does not happen with the HD signal. Maybe it's a buffer that has to be filled?
best

closed 2016-06-22 by git-artes