git-artes/gr-isdbt #16: Dongle DVB-t RTL2832+RT820t ISDBT 1seg

andregcoimbra, 2016-06-27:
Hi,

I have a dongle DVB-t rtl2832+rt820t and was atempted use with 1seg isdb-t project but not got any parameters. The dongle work well with FM and GSM reception. I need some special atention do make work?

Best Regards

git-artes, 2016-06-29:
Hi, 
We'll be more than glad to help, but you should be more specific on the problem you're having. Did you got gnuradio installed? Did you got gr-isdbt compiled? Does gr-isdbt runs? Do you see the constellation? Do you have trouble obtaining a usable multimedia signal?
best
Federico

andregcoimbra, 2016-07-01:
Hi, Frederico,

Since my question I really improved my tests. I have changed the blocks to work with my dongle and the result are this parameters but the results are inconstant. Follow the prints.
best.
Andre G. Coimbra

[image]

 
[image]

git-artes, 2016-07-02:
Hi, 
It looks like you got a really poor SNR. The correlation (in the first picture) should have a much more clear "peak", and you got what looks like the QPSK (what's normally transmitted in a 1-seg signal) but it's really noisy. Try a different antenna, or different position. 
You should also make sure to configure the rest of the blocks (time deinterleaver and viterbi decoder notably) accordingly (else, the Reed-Solomon decoder will deem the bytes incorrectable and not output anything, thus resulting in a file with 0 byte). 
best

andregcoimbra, 2016-07-02:
Hi, Thanks a lot for your help. Glad your tip I had change the antenna and worked very well. Now I will enjoy the project. 
Best regards

[image]

git-artes, 2016-07-03:
Glad it worked. 
best

closed 2016-07-03 by git-artes