git-artes/gr-isdbt #17: PARAMETERS MEASUREMENTS

eaiglesiasr, 2016-07-28:
Hi Federico.
Sometime ago, I was working with fullseg_receiver_and_measurements.gcc in order to receive Transport Stream from local DTV channel  which achieved, I attached one image. 
[image]

However, so far I haven´t been able to achieve good measurements of SNR, freq_error, and samp_error as same as figure 8 and 9 of "Efectos del canal sobre la señal de ISDB-T
Un caso de uso de gr-isdbt" paper. I attached two images.
[image]
[image]
Is it OK the way as I configured the before images? and 
Is it possible, Federico, that you can share  how I can configure in the good way the blocks to obtain SNR, freq_error, and samp_error, measurements,  please?

git-artes, 2016-07-28:
Hi, 
What do you mean by "I haven´t been able to achieve good measurements of SNR, freq_error, and samp_error"? You are not obtaining good measurements, or do you believe the measurements that gr-isdbt (or gr-mer) is reporting are wrong? 
From the graphs you are showing (specially the last one), it looks like you are getting a very low signal (you should get a much more triangular "peak"). You should try getting either a new antenna or a better position. Note that our report was made with a line-of-sight. When indoors, we tend to use an antenna specifically for DTV, or else we get similar measurements as what you are showing. 
best
Federico

eaiglesiasr, 2016-07-28:
Hi,
I mean by "I haven´t been able to achieve good measurements of SNR, freq_error, and samp_error" that  SNR measure, in particular, has been too unstable like it's possible to appreciate in the following figures:
[image]
[image]
[image]

git-artes, 2016-08-01:
Hi, 
The SNR estimation is taken from the estimation of the symbol alignment (this is described in the original paper the synchro algorithm is based on [1]). In previous picture you've shown a very noisy correlation function, which will result in unstable symbol alignment estimations, and thus SNR estimations. 
For more details you should refer to the article [1], and take a look at the Sym Align Ofdm internals. 
best
Federico
[1] J.-J. van de Beek, M. Sandell, and P. Borjesson, “ML estimation of time and frequency offset in OFDM systems,” Signal Processing, IEEE Transactions on, vol. 45, no. 7, pp. 1800–1805, Jul 1997

closed 2016-11-09 by git-artes