git-artes/gr-isdbt #18: error: cannot convert ‘unsigned int*’ to ‘uint16_t*

pqyptixa, 2016-07-30:
Hi.
I was getting this error when compiling:

    [...] error: cannot convert ‘unsigned int*’ to ‘uint16_t* {aka short unsigned int*}’ in argument passing [...]

Apparently, there were some changes in newer versions of volk: https://github.com/gnuradio/volk/issues/84

I made a quick patch to solve this issue:

```
diff --git a/lib/ofdm_sym_acquisition_impl.cc b/lib/ofdm_sym_acquisition_impl.cc
index d0213ed..a7b71cf 100644
--- a/lib/ofdm_sym_acquisition_impl.cc
+++ b/lib/ofdm_sym_acquisition_impl.cc
@@ -62,7 +62,7 @@ namespace gr {
         int 
             ofdm_sym_acquisition_impl::peak_detect_process(const float * datain, const int datain_length, int * peak_pos)
             {
-                unsigned int peak_index = 0;
+                uint16_t peak_index = 0;
                 bool success = true;
 
                 volk_32f_index_max_16u(&peak_index, &datain[0], datain_length); 
diff --git a/lib/ofdm_synchronization_1seg_impl.cc b/lib/ofdm_synchronization_1seg_impl.cc
index 2f68cc8..bd19252 100644
--- a/lib/ofdm_synchronization_1seg_impl.cc
+++ b/lib/ofdm_synchronization_1seg_impl.cc
@@ -641,7 +641,7 @@ namespace gr {
         bool
             ofdm_synchronization_1seg_impl::peak_detect_process(const float * datain, const int datain_length, int * peak_pos)
             {
-                unsigned int peak_index = 0;
+                uint16_t peak_index = 0;
                 bool success = true;
 
                 volk_32f_index_max_16u(&peak_index, &datain[0], datain_length); 
diff --git a/lib/ofdm_synchronization_impl.cc b/lib/ofdm_synchronization_impl.cc
index 764ac4c..f975b70 100644
--- a/lib/ofdm_synchronization_impl.cc
+++ b/lib/ofdm_synchronization_impl.cc
@@ -613,7 +613,7 @@ namespace gr {
         bool
             ofdm_synchronization_impl::peak_detect_process(const float * datain, const int datain_length, int * peak_pos)
             {
-                unsigned int peak_index = 0;
+                uint16_t peak_index = 0;
                 bool success = true;
 
                 volk_32f_index_max_16u(&peak_index, &datain[0], datain_length); 

```

I didn't check if the patch could be problematic, though... Anyway, I hope this helps.
Thanks for your work.

git-artes, 2016-08-01:
Thank you very much for the heads up! However, the solution is not easy. Naturally, if I change the type of peak_index to uint16_t, it won't compile with previous versions of VOLK (it won't compile with my GNU Radio 3.7.9.2). I'll dig a little bit to see how this may be fixed. 
best
Federico

pqyptixa, 2016-08-01:
There is a patch here that could be of help, I suppose:  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=828034#19

git-artes, 2016-08-01:
I guess something like that should work, but it will require some time to adapt to gr-isdbt and test (my knowledge con cmake is almost null). For the moment I'll post a warning on the README. 
Thanks!

ihab9187, 2016-08-04:
Don't understand. should be done to over come this problem? 
I mean if we need to make some changes what are they ? and where should they be applied? and to which file ?

ihab

git-artes, 2016-08-05:
Hi Ihab, 
New versions of VOLK has changed the data type in the function volk_32f_index_max_16u. It now requires a uint16_t instead of an unsigned int. I'm not sure why. 
OOTs modules like gr-isbt cannot simply change how they call the function, since we cannot assume a given version of VOLK. Thus, as suggested by @pqyptixa, we have to make some clever use of cmake and compiler instructions so that different things are compiled depending on the version of VOLK installed. 
We are starting the semester here, and I don't have much time, so I won't be able to take care of this in the short-term (pull-requests on this matter are welcome, by the way). In the meantime, if gr-isdbt does not compile and outputs an error as described in the subject of this issue, you should simply make the changes as instructed in the first message of this thread (in a nutshell, change three files so that volk_32f_index_max_16u uses the correct data type: lib/ofdm_sym_acquisition_impl.cc, lib/ofdm_synchronization_1seg_impl.cc and lib/ofdm_synchronization_impl.cc). 
best
Federico

ihab9187, 2016-08-05:
Hi Federico ,
Yeah I tracked the files this morning and I have fixed, it is installed and
working now.

Thanks
Ihab

-- 

This email originated from DIT. If you received this email in error, please 
delete it from your system. Please note that if you are not the named 
addressee, disclosing, copying, distributing or taking any action based on 
the contents of this email or attachments is prohibited. www.dit.ie

Is ó ITBÁC a tháinig an ríomhphost seo. Má fuair tú an ríomhphost seo trí 
earráid, scrios de do chóras é le do thoil. Tabhair ar aird, mura tú an 
seolaí ainmnithe, go bhfuil dianchosc ar aon nochtadh, aon chóipeáil, aon 
dáileadh nó ar aon ghníomh a dhéanfar bunaithe ar an ábhar atá sa 
ríomhphost nó sna hiatáin seo. www.dit.ie

Tá ITBÁC ag aistriú go Gráinseach Ghormáin – DIT is on the move to 
Grangegorman <http://www.dit.ie/grangegorman>

git-artes, 2016-11-09:
Hi @pqyptixa, 
The semester is ending, so I got a little time to work on this. I THINK I got it fixed. Let me know if it works for you. 
best
Federico

pqyptixa, 2016-11-27:
Hi @git-artes .
I cloned the repo and built the package successfully (using gcc 5.4.0 on Kubuntu 16.04.2)

closed 2016-11-28 by git-artes

miaoski, 2019-10-06:
The same issue happens in Ubuntu 18.04.3 and Volk 1.4.  Investigating.

miaoski, 2019-10-06:
It is quite interesting that `CMakeCache.txt` shows `PC_GNURADIO_VOLK_VERSION:INTERNAL=1.4`.  However, VOLK_GT_122 is not defined.  After tweaking ../lib/CMakeLists.txt (by chaging the default VOLK_VERSION to "1.4.0"), everything works.