git-artes/gr-isdbt #19: ISDBT SIGNAL RECEPTION PROBLEMS WITH USRP 2920 UHD: USRP SOURCE

Jordyggg, 2017-08-31:
Hi Federico, as I tell you, I'm working with the usrp 2920, and gr-isdbt works perfectly for me, when I work with a recorded signal, that is when I use the GNU radio file source and throttle blocks.

But when I enable the UHD blocks: USRP SOURCE and RATIONAL RESAMPLER (I assure you this is the frequency of the IFFT), to acquire the data of the signal with usrp, I can not retrieve the video, nor that of a segment. I tell you that I am using a transmitter of my own and I get a clear signal from isdbt above -50 dBm, also record my own signal in .dat and if I can recover the video, then I wanted to ask you. Why do I recover the video with a recorded signal and not when I use USRP ?, and what equipment did you use? Did they have the same problem?

Thanks for the reply

git-artes, 2017-08-31:
Hi, 
We used several SDR hardware, but mostly a USRP b100. They all worked, but as you mentioned, the sampling rate has to be properly set. If you are sure you've set that right, then a couple of problems and questions come to mind: 
 - the whole processing online is somewhat demanding. are you sure you are not dropping too many samples? 
 - do you get a "nice" constellation when using the USRP? I mean, did you check an intermediate stage in the receiver, or only the video?
best

Jordyggg, 2017-09-04:
- Yes i am. I consider that packets are lost, but not many.
-yes i do. I have checked intermediate stages the constellation is clean in the different layers of isdbt with the USRP, similar to those recorded by you.

The problem is that when I do not recover the video, it happens that the van de beek algorithm does not get static peaks, therefore I do not get the TMCC information, this I first check with obtain_parametres.grc. However as I mentioned  the constellation is clean, What do you think is the cause?

thanks for your help

git-artes, 2017-09-05:
Hi, 
If you are not getting the TMCC then you are probably dropping too many samples (and you are thus probably not setting correctly the time interleaver length and the coding). Could you post a picture of the constellation you are getting? And what is written on the terminal?
best

git-artes, 2017-10-24:
Since we didn't get any more feedback in almost a couple of months, I'll close this issue. You are free to submit a new one if you are having any further troubles.

closed 2017-10-24 by git-artes