git-artes/gr-isdbt #21: PMT linker issue ?

jmfriedt, 2017-10-23:
For what it's worth, while gr-isdbt compiles fine on a current (stretch) stable Debian system running gnuradio 3.7.10.1, it fails on a sid distribution with gnuradio 3.7.11.2 with
[ 94%] Linking CXX executable test-isdbt
libgnuradio-isdbt.so: undefined reference to `pmt::PMT_F'
both systems have gnuradio-dev installed.

git-artes, 2017-10-24:
Thanks for the head up. However, we focus on maintaining the project on the stable releases of OS and GNU Radio. BTW, did you mean 3.7.10.2? Because I think the current release are 3.7.10.2 and 3.7.11 (on which I have not yet tested gr-isdbt).

jmfriedt, 2017-10-27:
dpkg -l tells me I have 3.7.10.1 installed on Debian/stable. But forget my comment: the conflict was with a leftover lib in my /usr/local, your program compiles perfectly well on Debian/testing and sid, my mistake.
Thanks for the great gnuradio module.

closed 2017-10-27 by jmfriedt