git-artes/gr-isdbt #24: RuntimeError when running receiver grcs: block::set_relative_rate: relative rate must be > 0.0

TheRealPhoneCall, 2018-09-08:
Hi,
I wanted to try and test your module, and receive tv broadcast from my end. Was able to obtain parameters for a specific channel. But I'm having runtime error when running the following GRCs:

- fullseg_receiver_and_measurements.grc
- rx_demo.grc
- rx_1seg_demo.grc

Can you please help me move on from this error? Exact error stack trace is shown below...

Detected ISDB-T's 64QAM constellation
Detected ISDB-T's 64QAM constellation
[gr-isdbt.OFDM_SYM_ACQ] You are using ofdm_sym_acquisition to perform coarse OFDM synchronization. Consider using ofdm_synchronization instead. 
Traceback (most recent call last):
  File "/home/ergwd/Projects/gr-miindtv/examples/fullseg_receiver_and_measurements.py", line 580, in <module>
    main()
  File "/home/ergwd/Projects/gr-miindtv/examples/fullseg_receiver_and_measurements.py", line 568, in main
    tb = top_block_cls()
  File "/home/ergwd/Projects/gr-miindtv/examples/fullseg_receiver_and_measurements.py", line 421, in __init__
    rate=2,
  File "/home/ergwd/prefix/default/lib/python2.7/dist-packages/isdbt/isdbt_channel_decoding.py", line 41, in __init__
    self.isdbt_viterbi_decoder_0 = isdbt.viterbi_decoder(constellation_size, rate)
  File "/home/ergwd/prefix/default/lib/python2.7/dist-packages/isdbt/isdbt_swig.py", line 5043, in make
    return _isdbt_swig.viterbi_decoder_make(*args, **kwargs)
RuntimeError: block::set_relative_rate: relative rate must be > 0.0

git-artes, 2018-09-14:
Hi, 
That's a weird error. Those examples should work, and from what I understand, the error is regarding the set_relative_rate in the viterbi decoder. However, the variables involved are automatically set (depending on the options available), so it should definitely work. Maybe you could debug a little bit (like printing what's happening just before line 131 on viterbi_decoder_impl.cc)?
best

git-artes, 2018-10-03:
Hi, 
I meant printing the involved variables, like in printf("d_k: %i\n", d_k); to see what's going on with the variables involved in your particular case. 
best

ergwd-pcieerd, 2018-10-22:
Hi,
I'm the same person as the above but on another account. I got it running on another computer without that problem. It seems that there's some platform/dependency problem on my other computer. I will get back at you about it when I get back at that gnuradio workstation.

I have to ask, how do I open a video stream of the tv signals being received? Can you point me to a guide that does so? I might have to ask this on another thread

git-artes, 2018-10-22:
Hi, 
Glad it worked. 
To display the audio/video in the computer you should read the output file with a media player (we use ffplay). There is an explanation in the README. If you still have problems, please open another issue. 
best

closed 2018-10-22 by git-artes