git-artes/gr-isdbt #25: Command to use ffplay on test_output.ts file

ergwd-pcieerd, 2018-10-23:
Hi,
I just need to know what actual command to use in playing test_output.ts file with ffplay? I never used ffmpeg and ffplay before and directly using `ffplay test_output.ts` doesn't show anything. Do I need to convert the file with ffmpeg to a .mp4 format, then use ffplay? Just got to ask as this might actually be a one liner you guys readily know...

Thanks a lot for attending to my newbie questions.

Cheers,
Daryl

git-artes, 2018-10-23:
Hi, 
What's the size of the resulting test_output.ts? Because if it is 0 byte, it means the RS decoder is considering the stream beyond correction, typically meaning (if your constellation is "nice") that you are setting the parameters wrong (e..g time interleaver depth).
best

ergwd-pcieerd, 2018-10-27:
I see... I got ffplay running by using one of your demo .dat files (569MHz_recording.dat). It's constellation and video playback is shown in the attached image. So for it to be accepted by the RS decoder, I guess my constellation will have to like something like it right?

[image]

Will check test_out.ts on my own experiment. Will get back at this.

ergwd-pcieerd, 2018-10-27:
Going back to my own experiment, and checking test_out.ts, it is 0 byte indeed. The constellation was also nothing like in above (image below).
[image]

I'm attaching my flowgraph below, perhaps you can see which parameter I got wrong? I really appreciate you helping out on this.
[image]

ergwd-pcieerd, 2018-10-27:
Following up:

Is this the expected form of the obtaining parameters flowgraph?
[image]

I now understand the instructions better...but, I will need to find the parameters that can give this kind of form. 

Thanks,
Daryl

ergwd-pcieerd, 2018-10-29:
OK, thanks for confirming that...I will find those parameters - I'll get back at this on my next free time. You can close this issue now sir. as ffplay turned out fine when fed with correct isdbt data.

closed 2018-10-29 by git-artes