git-artes/gr-isdbt #28: Block key "isdbt_bit_deinterleaver" not found

Hijiki1127, 2019-01-26:
Hello,I am a student who recently started using GNUradio on MacOS. I installed GNUradio on my computer and tried to open rx_1seg_demo. However it did not work and it says missing block like the picture I attached. I followed you instructions on README and I thought installations were successful(I did not see any errors).  Could you help me to solve this problem?

Thank you.
[image]

Hijiki1127, 2019-01-27:
I now directly copied files in block folder and it does not show missing blocks. But it still has error which says

 Traceback (most recent call last):
  File "/Users/hibikishiiba/gr-isdbt/examples/rx_demo.py", line 29, in <module>
    import isdbt
ImportError: No module named isdbt

vagonbar, 2019-01-28:
>
> File "/Users/hibikishiiba/gr-isdbt/examples/rx_demo.py", line 29, in
> import isdbt
> ImportError: No module named isdbt
>
Please make sure your PYTHONPATH variable contains directory where isdbt
module is placed.
After you adjust your PYTHONPATH variable and duly export it, you may test
module isdbt import by invoking python in a terminal or a Python IDE like
Idle, and trying the command
  import isdbt
It should do the import with no error messages.

Hijiki1127, 2019-01-28:
Thank you for the comment.
Does it mean I have to open a path for the folder which is named isdbt?
And Should i type “import isdbt” after that?

If you don’t mind could you show the commands for me?

vagonbar, 2019-01-28:
Task: ensure your PYTHONPATH variable contains the directory where your
isdbt module can be found.

1) Verify is this is really your problem. The command
   echo $PYTHONPATH
should contain the directory where your isdbt module is placed. If the
directory is present, this is OK and the origin of your problem must be
sought elsewhere.

2) If you did a normal GNU Radio installation, you can try this:
  source <full path to your GNU Radio installation>/setup_env.sh
You may find your setup_env.sh in your machine and use the full path
obtained.
After this, step 1) should show all paths to your GNU Radio installation,
including the one where isdbt can be found.
  If the command was successful, you may try to invoke GNU Radio Companion
and see if it works with isdbt.
  If this worked, you may include the "source" command at the end of your
.bashrc for it to be executed each time you open a terminal.

3)  More specifically, you can test isdbt import by issuing the following
commands in a Linux terminal:
    python  # opens a Python command line terminal
    import isdbt
It should silently return. The command
    dir(isdbt)
should show all components of the isdbt module. Write exit() or Ctrl-D to
end the Python session.

git-artes, 2019-02-01:
Hi, 
If you had to copy the grc files into the folder it means that something went wrong during the installation process. Are you sure you had administrative privileges when you issued "sudo make install"??
best

Hijiki1127, 2019-03-07:
Hi there, thank you for the response.
 When I execute the command sudo make install, terminal shows a result below.

$ make && sudo make install
[  1%] Linking CXX shared library libgnuradio-isdbt.dylib
[ 41%] Built target gnuradio-isdbt
[ 43%] Linking CXX executable test-isdbt
[ 71%] Built target test-isdbt
[ 72%] Linking CXX executable _isdbt_swig_doc_tag
[ 74%] Built target _isdbt_swig_doc_tag
[ 75%] Generating doxygen xml for isdbt_swig_doc docs
warning: Tag `XML_SCHEMA' at line 1478 of file `/Users/hibikishiiba/gr-isdbt/build/swig/isdbt_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `XML_DTD' at line 1484 of file `/Users/hibikishiiba/gr-isdbt/build/swig/isdbt_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[ 77%] Generating python docstrings for isdbt_swig_doc
[ 77%] Built target isdbt_swig_swig_doc
[ 78%] Linking CXX executable _isdbt_swig_swig_tag
[ 79%] Built target _isdbt_swig_swig_tag
[ 81%] Generating isdbt_swig.tag
Scanning dependencies of target isdbt_swig_swig_2d0df
[ 82%] Building CXX object swig/CMakeFiles/isdbt_swig_swig_2d0df.dir/isdbt_swig_swig_2d0df.cpp.o
[ 83%] Linking CXX executable isdbt_swig_swig_2d0df
Swig source
[ 86%] Built target isdbt_swig_swig_2d0df
Scanning dependencies of target _isdbt_swig
[ 87%] Building CXX object swig/CMakeFiles/_isdbt_swig.dir/isdbt_swigPYTHON_wrap.cxx.o
In file included from /Users/hibikishiiba/gr-isdbt/build/swig/isdbt_swigPYTHON_wrap.cxx:173:
In file included from /usr/include/python2.7/Python.h:85:
/usr/include/python2.7/unicodeobject.h:534:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register PyObject *obj,     /* Object */
    ^~~~~~~~~
/usr/include/python2.7/unicodeobject.h:553:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register PyObject *obj      /* Object */
    ^~~~~~~~~
/usr/include/python2.7/unicodeobject.h:575:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register const wchar_t *w,  /* wchar_t buffer */
    ^~~~~~~~~
/usr/include/python2.7/unicodeobject.h:593:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register wchar_t *w,        /* wchar_t buffer */
    ^~~~~~~~~
In file included from /Users/hibikishiiba/gr-isdbt/build/swig/isdbt_swigPYTHON_wrap.cxx:173:
In file included from /usr/include/python2.7/Python.h:94:
/usr/include/python2.7/stringobject.h:173:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register PyObject *obj,     /* string or Unicode object */
    ^~~~~~~~~
/usr/include/python2.7/stringobject.h:174:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register char **s,          /* pointer to buffer variable */
    ^~~~~~~~~
/usr/include/python2.7/stringobject.h:175:5: warning: 'register' storage class
      specifier is deprecated and incompatible with C++17
      [-Wdeprecated-register]
    register Py_ssize_t *len    /* pointer to length variable or NULL
    ^~~~~~~~~
7 warnings generated.
[ 89%] Linking CXX shared module _isdbt_swig.so
[ 91%] Built target _isdbt_swig
[ 93%] Generating isdbt_swig.pyc
[ 94%] Generating isdbt_swig.pyo
[ 95%] Built target pygen_swig_6e74a
[ 98%] Built target pygen_python_3f59a
[ 98%] Built target pygen_apps_9a6dd
[100%] Built target doxygen_target
Password:
[ 41%] Built target gnuradio-isdbt
[ 71%] Built target test-isdbt
[ 74%] Built target _isdbt_swig_doc_tag
[ 77%] Built target isdbt_swig_swig_doc
[ 79%] Built target _isdbt_swig_swig_tag
[ 86%] Built target isdbt_swig_swig_2d0df
[ 91%] Built target _isdbt_swig
[ 95%] Built target pygen_swig_6e74a
[ 98%] Built target pygen_python_3f59a
[ 98%] Built target pygen_apps_9a6dd
[100%] Built target doxygen_target
Install the project...
-- Install configuration: "Release"

Thank you.

git-artes, 2019-03-08:
I've just realized you are using Mac. Please note that our intended working environment is Linux, and we have not tested gr-isdbt on MacOS. The problem is probably there. You should check if there are any changes that need to be done to OOT as ours in order to work on Mac. 
best