git-artes/gr-isdbt #29: GNURadio on windows

rubenervedal, 2019-02-07:
Can this be used with windows gnuradio? I already test this project on linux, but now i want to use it on windows. I already have gnuradio installed on windows, but I can't building the project on windows. I already tried Cygwin and CMake withou success.

git-artes, 2019-02-07:
Hi, 
This was asked in issue #27 , but as I answered there, we have stick to linux in all our developments. We have thus no experience in working with GNU Radio on windows, let alone compiling an OOT. Maybe you could ask this question in the discuss-gnuradio list? After all, gr-isdbt does not have any external libraries, so you may ask in general how to compile an OOT on windows. 
best

closed 2019-02-07 by git-artes

git-artes, 2020-04-29:
Hi, 
In case you are still interested, I've checked the new WSL and gr-isdbt works. 
best