git-artes/gr-isdbt #30: Layer B cannot be displyed

fwkchenchen, 2019-02-14:
Hi, I am a student from Sendai, Japan. Thanks for providing this useful ISDB-T transceiver :)

I faced a problem with the video playing with the real-sampled data (the results of the 569MHz data is of course very good, no problem). 
I sampled the data by using B210 with a sampling rate of 512/63 MHz.
In Sendai, mode-3 is used and the CP length is 1/8.
I can demodulate the received signal and play the video of layer a (1 segment) by ffplay.
But, for layer b, the video cannot be played (I am not familiar with ffmpeg).

The errors are something in the following:
[mpegts @ 0x7fe3b4000b80] start time for stream 0 is not set in estimate_timings_from_pts
[mpegts @ 0x7fe3b4000b80] start time for stream 1 is not set in estimate_timings_from_pts
[mpegts @ 0x7fe3b4000b80] start time for stream 2 is not set in estimate_timings_from_pts
[mpegts @ 0x7fe3b4000b80] start time for stream 3 is not set in estimate_timings_from_pts
[mpegts @ 0x7fe3b4000b80] Could not find codec parameters for stream 0 (Video: mpeg2video ([2][0][0][0] / 0x0002), none(tv)): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
[mpegts @ 0x7fe3b4000b80] Could not find codec parameters for stream 1 (Audio: aac ([15][0][0][0] / 0x000F), 0 channels, fltp): unspecified sample rate
Consider increasing the value for the 'analyzeduration' and 'probesize' options

I think, for this problem, the number of segments and the segment length of layer B should be considered.
In your implementation, some of the parameters, such as CP length, can be somehow automatically obtained. How about the segments of layer A, B and C?
Could you give some suggestions?

Thanks and best regards,
Feng

git-artes, 2019-02-14:
Hi, 
There are two possible problems: 
 1 - Are you trying to decode both layers at the same time? In my experience, it is not possible (I'm not sure, but I think it's because of the implementation of the viterbi decoder). You have to decode one layer at a time. Feed layer A to a null sink as in the examples included. 
 2 - Have you correctly setup the parameters for the layer? You should get a relatively small BER on the RS decoder. The parameters may be obtained by choosing yes in "Print params" variable in the TMCC decoder block. 
Hope it helps!
best

fwkchenchen, 2019-02-14:
Hi, thanks for your comments.
1. Yes, I also found this problem and found that we should separately decode layer a and layer b. So, this may not be a problem.
2. Yes, I notice that we can check the parameters from the TMCC decoder and I get the layer segmentation setup (the parameters are all same to the reference 569 MHz data, expect the CP length). So, this may also not a problem.

We also tested mplayer (which is unable to play layer a and layer b) and vlc, which can also play layer a but provides the following errors on layer b

[00007f2090c18090] ts demux error: libdvbpsi error (PSI decoder): TS 
discontinuity (received 5, expected 15) for PID 0
[00007f2090c18090] ts demux error: libdvbpsi error (PSI decoder): TS 
discontinuity (received 8, expected 2) for PID 257
[00007f2090c18090] ts demux error: libdvbpsi error (PSI decoder): TS 
discontinuity (received 8, expected 2) for PID 258

so I am afraid the error is in the demodulation. VLC shows the time bar
increasing as it is reading the file, but cannot display any picture.

Thanks for your help!

fwkchenchen, 2019-02-14:
By the way, the measurements of reception is as follows:

It seems that the Viterbi BER is not so good.
[image]

best

git-artes, 2019-02-14:
Mmmmm this means that you have misconfigured some of the parameters of layer B. With such a constellation you should get a very low Viterbi BER (the one you are getting now is as if you've input random bits). 
You should double-check the parameters of the layer: interleaving length, constellation type, and coding rate. 
Maybe you could enable print params in the TMCC decode block and share with us a screen capture of the flowgraph?

fwkchenchen, 2019-02-15:
Yes, I double-checked the layer parameters. It seems there is no mistake. Two flowgraphs I used are as follows:
[image]
[image]

thanks

git-artes, 2019-02-15:
You seem to be right. Could you share with us the recording for us to check?

fwkchenchen, 2019-02-15:
Yes, thanks so much for your help.
Since the recorded data file is big. I am sending to you by email.

git-artes, 2019-02-15:
I've just sent you a mail, but just to be "public" with the answer, my suspicion is that the problem is with ffmpeg and the encoding used on the Transport Stream on layer B. We'll investigate a little bit further and try to confirm this. 
In the meantime, another thing you may try is checking whether the problem is with this particular station (channel) or if you are able to correctly decode all layers of other stations.

fwkchenchen, 2019-02-15:
Yes, thanks for the help and I got your email.

Actually, I have sampled three channels data from the same TV tower (in Sendai, there all totally three TV towers). For the sampled three channels, only layer a can be displayed.
I will try to sampled some data from other two TV towers and to check the results.
I will let you know about the results.

Thanks.

git-artes, 2019-03-26:
Hi, 
Thanks to @pablofloresguridi we seem to have find the problem with your particular recording: layer B is encrypted. We've used DVBsnoop (http://dvbsnoop.sourceforge.net/), and got the following output (in bold the part that interests us): 

------------------------------------------------------------
TS-Packet: 00000001   PID: (Unkown PID), Length: 188 (0x00bc)
from file: salida_layer_b.ts
------------------------------------------------------------
  0000:  47 01 11 d1 0a 1e aa e3  32 20 18 4f 79 0f 8f fc   G.......2 .Oy...
  0010:  79 58 8f 69 b0 72 8e 78  1c d5 d5 ca 63 db d2 33   yX.i.r.x....c..3
  0020:  79 7d d4 13 2a 51 43 b8  5e ae 69 35 3e 33 b3 6b   y}..*QC.^.i5>3.k
  0030:  08 f6 9f 2f 6e 92 df 11  27 2d 2d 0b 92 8d 0b 83   .../n...'--.....
  0040:  58 67 67 42 f5 45 13 da  26 b7 67 85 20 26 6b 5b   XggB.E..&.g. &k[
  0050:  06 f4 ea 05 80 1e 96 31  d7 f3 45 f1 ad 5a f3 a2   .......1..E..Z..
  0060:  0f af 1d a6 c2 a2 0c 23  d1 e8 f0 96 72 4a ad 77   .......#....rJ.w
  0070:  df 47 f3 b3 38 a8 20 59  ce 72 44 de 23 c1 8e 8a   .G..8. Y.rD.#...
  0080:  ec 9b a4 59 8f 99 b6 d0  12 b0 b5 38 64 40 45 c6   ...Y.......8d@E.
  0090:  32 af 8c 44 03 ae 90 2f  86 fb 9d 2d 7e eb 86 ff   2..D.../...-~...
  00a0:  87 a4 3e b3 3d 5a 5a b4  c1 4b da 06 64 34 3d 29   ..>.=ZZ..K..d4=)
  00b0:  4c 0a 53 8d 40 cd 3a fc  ec 75 c4 28               L.S.@.:..u.(

Sync-Byte 0x47: 71 (0x47)
Transport_error_indicator: 0 (0x00)  [= packet ok]
Payload_unit_start_indicator: 0 (0x00)  [= Packet data continues]
transport_priority: 0 (0x00)
PID: 273 (0x0111)  [= ]
**transport_scrambling_control: 3 (0x03)  [= TS packet scrambled with Odd Key]**
adaptation_field_control: 1 (0x01)  [= no adaptation_field, payload only]
continuity_counter: 1 (0x01)  [= (sequence ok)]
    Payload: (len: 184)
    Data-Bytes:
          0000:  0a 1e aa e3 32 20 18 4f  79 0f 8f fc 79 58 8f 69   ....2 .Oy...yX.i
          0010:  b0 72 8e 78 1c d5 d5 ca  63 db d2 33 79 7d d4 13   .r.x....c..3y}..
          0020:  2a 51 43 b8 5e ae 69 35  3e 33 b3 6b 08 f6 9f 2f   *QC.^.i5>3.k.../
          0030:  6e 92 df 11 27 2d 2d 0b  92 8d 0b 83 58 67 67 42   n...'--.....XggB
          0040:  f5 45 13 da 26 b7 67 85  20 26 6b 5b 06 f4 ea 05   .E..&.g. &k[....
          0050:  80 1e 96 31 d7 f3 45 f1  ad 5a f3 a2 0f af 1d a6   ...1..E..Z......
          0060:  c2 a2 0c 23 d1 e8 f0 96  72 4a ad 77 df 47 f3 b3   ...#....rJ.w.G..
          0070:  38 a8 20 59 ce 72 44 de  23 c1 8e 8a ec 9b a4 59   8. Y.rD.#......Y
          0080:  8f 99 b6 d0 12 b0 b5 38  64 40 45 c6 32 af 8c 44   .......8d@E.2..D
          0090:  03 ae 90 2f 86 fb 9d 2d  7e eb 86 ff 87 a4 3e b3   .../...-~.....>.
          00a0:  3d 5a 5a b4 c1 4b da 06  64 34 3d 29 4c 0a 53 8d   =ZZ..K..d4=)L.S.
          00b0:  40 cd 3a fc ec 75 c4 28                            @.:..u.(
==========================================================

Hope it helps!
best
Federico

mmmdg, 2019-05-22:
> By the way, the measurements of reception is as follows:
> 
> It seems that the Viterbi BER is not so good.
> [image]
> 
> best

Hi,
I am also exploring this ISDB-t implementation for GNURadio. I would like to ask if how you were able to produce/measure the BER information?

Thanks