git-artes/gr-isdbt #31: RPi3 Installation Error

mmmdg, 2019-04-24:
I am trying to install the library to a raspberry pi 3 model B machine. However, it seems that the source code does not handle arm devices yet. Upon running the make command, I encountered the following error:

```
[ 12%] Building C object lib/CMakeFiles/gnuradio-isdbt.dir/d_viterbi.c.o
In file included from /home/m/gr-isdbt/lib/d_viterbi.c:32:0:
/home/m/gr-isdbt/lib/d_viterbi.h:29:23: fatal error: xmmintrin.h: No such file or directory
compilation terminated.
```
Based on my initial research, xmmintrin.h does not support arm devices. Can you recommend any workaround on this issue? Or do you have plans in the future to support arm devices?

git-artes, 2019-04-24:
Hi, 
Thanks for the heads up, as we are planning to start working on the raspberry pi shortly (however, don't expect any changes on the code before some months). You may wanna take a look at gr-dtv, which has several compiler instructions to fallback to a generic implementation of the viterbi decoder. In fact, our decoder is mostly the same, except that ours output the BER (and expects a certain tag indicating when to reset). 
Note however that I don't think you will be able to decoder the whole signal. Maybe the 1-seg segment?
best
PS Pull Requests are always welcome :-)

mmmdg, 2019-04-25:
> Hi,
> Thanks for the heads up, as we are planning to start working on the raspberry pi shortly (however, don't expect any changes on the code before some months). You may wanna take a look at gr-dtv, [...]

Thanks :)

closed 2019-04-25 by mmmdg