git-artes/gr-isdbt #32: Please upload testing data

miaoski, 2019-10-02:
Hi.
Thank you for this module!  I live in a country where there is no ISDB-T, but I got an ISDB-T set-top box from a neighbor country.  Is it possible to upload testing data, so I can play with your module on my BladeRF?
Thank you!

git-artes, 2019-10-02:
Hi, 
Thanks for the feedback. You may get several recordings from the project's webpage: http://iie.fing.edu.uy/investigacion/grupos/artes/gr-isdbt/ (there were some problems yesterday and today so it may not be up, but it will be very shortly). 
There are over-the-air recordings (to test with the receiver) and a couple of TS files to test with the transmitter (they are the ones to be used with the tx_demo.grc flowgraph). 
Let us know how it worked for you.
best

miaoski, 2019-10-02:
Hi.
I got 503 access denied several hours ago.  Now it's working.  I'm downloading the samples.  Thank you!

closed 2019-10-02 by miaoski