git-artes/gr-isdbt #35: Captured signal in the Philippines

miaoski, 2019-10-09:
Hi.
This is the best signal I can get in the Philippines.  I see clear 16QAM (layer B) + QPSK (layer A).  Feel free to share it with people / on your website / on GitHub, etc.

http://ckhis.ck.tp.edu.tw/~ljm/isdb-t/manila-569143000.dat.gz

I captured it with BladeRF x45, sample rate 8 MHz.

miaoski, 2019-10-09:
I have also tried to capture OTA firmware upgrade (617.143 MHz, Download PID 0x0EDF, Table ID 0x9C).  The signal is good for the set-top box, but bad for BladeRF (my antenna does not fit the range.)

tech-user-1, 2020-12-03:
Hello,

The link is down. Can you please upload the recording again?

Thank you very much!

miaoski, 2021-03-27:
Uploaded it to Google Drive: https://drive.google.com/file/d/103dR3SRJsx0xaM38gt3TFxtHNVqHAHH6/view?usp=sharing
If it doesn't work, please tell me.