git-artes/gr-isdbt #38: Error: Block key "blocks_multiply_const_xx" not found

CristianMaza, 2019-10-28:
Hi. I'm trying to execute the example but it doesn't work.
I'm working on Ubuntu 16.04 with GNU Radio Companion 3.7.9.
The tx_demo.grc file has a missing block as you can see in the following image:

[image]

What is the missing block? 

The full_transceiver.grc file doesn't have any block error but when I execute it, it shows this error:
 
[image]

How can I fix it?
Thank in advance.

git-artes, 2019-10-28:
Hi, 
You seem to have a problem on your GNU Radio installation: the missing block is multiply constant, which should be installed. You are also using a very old version of GNU Radio (over three years old). Maybe that's the problem?
best

CristianMaza, 2019-10-31:
Thank you for your answer. Yes that was the problem. I tried to upgrade the version of GNU Radio in Ubuntu 16 but it was difficult due multiple bugs and dependencies. The fast solution was install Ubuntu 18 instead of Ubuntu 16. Use apt-get to install GNU Radio, it will install the upgraded version. I tried to install the isdb-t library and it worked fine!

closed 2019-10-31 by CristianMaza