git-artes/gr-isdbt #40: SDR Transmitter to TV Receiver Test

mmmdg, 2020-05-08:
I have not successfully able to transmit TS file from SDR Transmitter to TV Receiver using the gr-isdbt transmitter.
The following are the tests we have conducted.

[image]

[image]

Note that the IQ file mentioned above is the 569MHz_recording.tar.gz from your website. I also tried replacing it with my own IQ recording from TV transmitters in our location. 
In addition, to verify that the New IQ File works, I tried decoding it using the flow New IQ File -> GR-RX -> TS File. The TS file generated from this test is working. So now it confuses me where the problem could come from since this series of tests sort of verified that the GR-TX, and GR-RX are both working.
 
I would like to confirm if you were able to successfully transmit from ISDBT GR-TX to a commercial TV receiver?

Thank you.

git-artes, 2020-05-08:
Hi, 
Yes, the transmitter works with commercial TVs, although it depends on the TV how it reacts to lost samples on the transmitting SDR. You should check this is not happening. I'm assuming you have already tested using an SDR as receiver and that it worked. In that case you may also use the receiving SDR for strange patterns on the signal (such as the aforementioned lost samples, evidenced by "hops" on the Viterbi BER, or showing the received complex symbols).
In any case, what does "not working" means exactly here? Does the TV shows nothing at all? Does it shows intermittent images?
What is the configuration of the transmitter? What TSs are you feeding it with? You may generate your own TS and send it over the air by following the tutorial on our website (https://iie.fing.edu.uy/investigacion/grupos/artes/projects/gr-isdbt/ at the "Documents" section). 
best

mmmdg, 2020-05-09:
Hello,

**Working** means that TV program was successfully added to the receiver's list of program and I can watch the video. This happens when I simply transmit the raw IQ recording from your source and from my own recording. The quality is also good and there were almost no dropped images.

**Not working** means that no TV program was added to the list and I cannot watch the video. This happens when I generate IQ file using the GR-TX. 

This is the TV screen during manual search:
[image]
During the processing of the channel which lasts around 7 seconds, the strength & quality shows a number and appeared to be updated at a rate of once or twice per second.
In the Working case, the strength = 55 and quality = 99. These values appear 90% of the time during the processing and both do not show any value 10% of the time.
In the Not Working case, almost same values were displayed but only appear 50% of the time.

From the images I uploaded in my first comment, I generated the TS file (using GR-RX) from your sample IQ recording as well as from my own recording. That TS file seems to be working since I can play it in VLC. Another clarification on the images above, I conducted 4 different kinds of tests in total. The labels "1", and "2" simply indicate the step number in a particular test.

This is the GR-RX used to convert raw IQ file to TS file. This is the same GR-RX I used to successfully receive from TV to SDR.
[image]
This is the GR-TX.
[image]
This is the GR IQ TX or the flowgraph I used to transmit the IQ file to the TV transmitter
[image]

git-artes, 2020-05-09:
I would first try following the tutorial on the webpage and see if that works. Let me know how that works out.
best

mmmdg, 2020-05-10:
I tried the instructions from transmitting_webcam_or_videos.pdf document.
The only parameters that I modified are the file source and center_freq. Since I did not modify any of the transmission parameters: mode, guard interval, code rate, etc., I generated the ts file by using the ffmpeg command given in the instructions. I only changed the source and destination filenames.

Upon running the flowgraph, the TV receiver failed to display the video again. The response is similar to what I described before regarding the Not Working case. The Strength and Quality readings are intermittent, and the total TV added is Zero.

BTW, I noticed that there are some warnings during runtime:

`-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Asking for clock rate 32.507936 MHz... 
-- Actually got clock rate 32.507936 MHz.
-- Performing timer loopback test... pass
tsp_per_ofdm_frame: 832
d_tsp_per_frame: 832
extra_delay in tsps: 821
gr::buffer::allocate_buffer: warning: tried to allocate
   321 items of size 204. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   321 items of size 204. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   10 items of size 6048. Due to alignment requirements
   128 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 2000000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   348 items of size 188. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
-- Asking for clock rate 32.507936 MHz... OK
-- Asking for clock rate 32.507936 MHz... OK`

mmmdg, 2020-05-26:
I reviewed the gr-isdbt readme and I realized that it is the "tx_demo.grc" flowgraph that you have confirmed working in viewing the video on commercial TVs. I was also able to verified that this approach works in our setup. I used the flowgraph with inputs "layer_a.ts" and "layer_b.ts" from your website.

This is exactly what we wanted to do. However, when I attempt to use the ts file that I generated using the approach from this https://iie.fing.edu.uy/investigacion/grupos/artes/wp-content/uploads/sites/13/2019/12/transmitting_webcam_or_videos.pdf, we cannot successfully display the video on the TV.
Perhaps giving us information on how you generated these files ("layer_a.ts" and "layer_b.ts") could help us to move forward.

Thanks

mmmdg, 2020-05-26:
It works now. I used tx_demo.grc instead of tx_demo_13seg. I used null source at layer a, and my own video at layer b.

git-artes, 2020-05-26:
Glad it worked. Weird that you needed to set layer_a as nulls, since it should not match the bitrate. 
I'll close this issue anyway, but let me know if you have further questions.
best

closed 2020-05-26 by git-artes