git-artes/gr-isdbt #42: Weird BER Measurements

ceeae10, 2020-05-27:
I'm varying the transmitter and receiver gain to show the differences in signal power, constellation clarity, MER measurements, and BER Measurements. However, the BER Measurements I have been getting is at -18dB. I have read that the expected value for BER is at 2x10^-4(around -3.698dB). Is this an error? Because I find having an -18 dB hard to believe.

Thanks for your help

git-artes, 2020-05-27:
Hi, If you are getting -18 dB it means that you have zero error (not sure why, probably internally GNU Radio adds a little epsilon to the log). Is this is a simulation? Is it the Viterbi or Reed-Solomon BER?
best

ceeae10, 2020-05-28:
This is wireless transmission. It's the Viterbi BER. The change was drastic from around ~-3dB at lower signal power and then when increased, it shoots up to -18dB. From there, it randomly fluctuates. See the picture for reference. 

This is RX Gain at 30 dB and TX Gain at 65 dB
[Reference](https://drive.google.com/file/d/15ZRgwZZNGTDjduGde-JLbVFtJqudSx3Z/view?usp=sharing}

git-artes, 2020-05-28:
Then you are getting zero error during your high-power transmission. The recording we share at the project's website has a BER of 10^(-9) or less, as we had line-of-sight with station's antenna. In this case, where I am assume you are transmitting next to the receiver, it's not rare that you get a virtually flawless signal.