git-artes/gr-isdbt #46: Fail to build: tmcc_decoder_impl has no member named ‘get_tags_in_window’

LeonardoSanBenitez, 2021-04-06:
Hello,
I'm trying to install the package, but I'm struggling with the build.

`cmake ../` seems to run well, the output is:
```
-- The CXX compiler identification is GNU 4.8.5
-- The C compiler identification is GNU 4.8.5
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'cppunit'
--   found cppunit, version 1.13.1
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
Checking for GNU Radio Module: RUNTIME
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.2
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so
-- Found GNURADIO_RUNTIME: /usr/local/lib/libgnuradio-runtime.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: FEC
-- checking for module 'gnuradio-fec'
--   found gnuradio-fec, version 3.7.2
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-fec.so
-- Found GNURADIO_FEC: /usr/local/lib/libgnuradio-fec.so  
GNURADIO_FEC_FOUND = TRUE
Checking for GNU Radio Module: FFT
-- checking for module 'gnuradio-fft'
--   found gnuradio-fft, version 3.7.2
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-fft.so
-- Found GNURADIO_FFT: /usr/local/lib/libgnuradio-fft.so  
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- checking for module 'gnuradio-filter'
--   found gnuradio-filter, version 3.7.2
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-filter.so
-- Found GNURADIO_FILTER: /usr/local/lib/libgnuradio-filter.so  
GNURADIO_FILTER_FOUND = TRUE
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.11") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.6", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.6", minimum required is "2") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: <path>/gr-isdbt/build
```

However, when I run `make`, I receive the error:
```
[  1%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o
[  3%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/sync_and_channel_estimation_impl.cc.o
[  4%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/subset_of_carriers_impl.cc.o
[  6%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/frequency_deinterleaver_impl.cc.o
[  7%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/time_deinterleaver_impl.cc.o
[  9%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/symbol_demapper_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/gnuradio-isdbt.dir/tmcc_decoder_impl.cc.o
/home/zads/Downloads/gr-isdbt/lib/tmcc_decoder_impl.cc: In member function ‘virtual int gr::isdbt::tmcc_decoder_impl::general_work(int, gr_vector_int&, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/zads/Downloads/gr-isdbt/lib/tmcc_decoder_impl.cc:681:27: error: ‘class gr::isdbt::tmcc_decoder_impl’ has no member named ‘get_tags_in_window’
                     this->get_tags_in_window(tags,0,i,i+1, pmt::string_to_symbol("resync"));
                           ^
/home/zads/Downloads/gr-isdbt/lib/tmcc_decoder_impl.cc:716:27: error: ‘class gr::isdbt::tmcc_decoder_impl’ has no member named ‘get_tags_in_window’
                     this->get_tags_in_window(tags,0,i,i+1, pmt::string_to_symbol("relative_symbol_index"));
                           ^
make[2]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/tmcc_decoder_impl.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/all] Error 2
make: *** [all] Error 2
```

------
Operational System: Ubuntu 14.04.1
Architecture: x86_64
cmake version 2.8.12.2
Running on a virtual machine VirtualBox 6.1.16
gnuradio version: 3.7.2
-------

I would appreciate any help, thank you in advance.

git-artes, 2021-04-06:
Hi, 
Any reason why you are using such old version of GNU Radio? You should use 3.7.11 or newer (although probably previous versions should work) in an Ubuntu 16 or newer. Note however that gr-isdbt is currently not compatible with GNU Radio 3.8. 
best

LeonardoSanBenitez, 2021-04-06:
I've tried with gnuradio version 3.7.11, but the compilation of gr-isdbt complains about the version of gnuradio.
I've commented out the line 129 CMakeLists.txt (`find_package(Gnuradio "3.7.2" REQUIRED)`) and tried to compile again, in an ubuntu 18.04 with gnuradio 3.7.11, but now I receive the following error after execute `make`:
```
gr-isdbt/lib/ofdm_sym_acquisition_impl.cc:35:10: fatal error: gnuradio/io_signature.h: No such file or directory
 #include <gnuradio/io_signature.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/gnuradio-isdbt.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/ofdm_sym_acquisition_impl.cc.o] Error 1
CMakeFiles/Makefile2:174: recipe for target 'lib/CMakeFiles/gnuradio-isdbt.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-isdbt.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

By the way, do you have a virtual machine with gr-isdbt already installed? If so, would you kindly make it available?

git-artes, 2021-04-07:
Hi, 
I don't have a virtual machine, but I can offer you a couple of pointers: 
 1 - I have a docker that should be able to install gr-isdbt (and other OOTs): https://github.com/git-artes/docker-gnuradio. Use the 3.7 version. 
 2 - If you have Windows, you may use WSL with an Ubuntu 18, which I've tested gr-isdbt on. 
best

LeonardoSanBenitez, 2021-04-08:
Thank you very much!

I successfully compiled gr-isdbt in this live image with gnu radio: https://www.gnuradio.org/news/2017-03-27-gnu-radio-release-3-7-11-and-3-7-10-2/
It has gnu radio 3.7.11 in an Ubuntu 16.04.

I'll leave the issue open, since the problems persist with the aforementioned configurations.

LeonardoSanBenitez, 2021-04-08:
Update:
I successfully compiled in ubuntu 18.04, I was just missing some packages. Everything worked after installing these (probably gnuradio-dev was the one missing):
`sudo apt install -y gnuradio gnuradio-dev libboost-all-dev libcppunit-dev liblog4cpp5-dev swig liborc-dev`

git-artes, 2021-04-08:
Great. I'll close this issue then. I think these requirements were listed on the README.
best

closed 2021-04-08 by git-artes