git-artes/gr-isdbt #47: tx_demo.grc not work on mobile devices!

CassianoDev, 2021-08-05:
I've tried everything and I can't get a cell phone to play the channel, I've tested one segment / 12 segments and nothing, all types of transmission, changed: rate, guard, modulation and etc and not work.

When searching, the cell phone stops for a few seconds detecting the signal like a TV does, but it after seconds continues to search without playing the transmission...

All TVs played both the full segment and the single segment, but the cell phone doesn't.
They seem to recognize the signal but something strange happens that they don't ring!

Is flowchart:
[image]

git-artes, 2021-08-05:
Thanks for the interest in the project. We've tested our system in TVs only. We've had feedback from people trying to use it on PSP decoders that couldn't make it work, and it seems to be a video encoding issue. I'm sorry to be of little help, but without the actual hardware to test it I'm not sure we may do much more.

CassianoDev, 2021-08-06:
> Thanks for the interest in the project. We've tested our system in TVs only. We've had feedback from people trying to use it on PSP decoders that couldn't make it work, and it seems to be a video [...]

Got it, I wanted it to work on the smartphone! :(
I don't know where to start to make it work.

CassianoDev, 2022-01-06:
After last update to Gnuradio 3.8, smartphone work now! 
Thanks

closed 2022-01-06 by CassianoDev