git-artes/gr-isdbt #49: problem to install module gr-isdbt on gnuradio 3.8

Samircaiala, 2022-06-05:
I'm trying to install the isdbt module on my gnuradio 3.8 version and when i perform the comand "make && sudo make install", i am geting the following ERROR:

"make[2]: *** Sem regra para processar o alvo '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', necessário por 'lib/libgnuradio-isdbt.so.1.0.0.0'.  Pare.
make[1]: *** [CMakeFiles/Makefile2:272: lib/CMakeFiles/gnuradio-isdbt.dir/all] Erro 2
make: *** [Makefile:141: all] Erro 2"

How can  I solve this Issue ? 

[image]

If it's possible, please answer to my request throgh the following E-mail adress: samir.engine@outlook.com

git-artes, 2022-06-06:
Hi, 
You don't have orc installed. You should first install all pre-requisites through: 
$ sudo apt-get install gnuradio-dev cmake git libboost-all-dev libcppunit-dev liblog4cpp5-dev swig liborc-dev libgsl-dev

and then the safest way is to erase the whole build directory and start again. 
best

Samircaiala, 2022-06-06:
@git-artes ,wow,thanks so much for this fast answer, i'm going to just like you said! 

best regards!

Samircaiala, 2022-06-28:
Thanks a lot! Problem solved , just by doing what you told me to do!

closed 2022-06-28 by Samircaiala