git-artes/gr-isdbt #50: ModuleNotFoundError: No module named 'isdbt'

Samircaiala, 2022-06-28:
Hello!! I am performing a graduation tesis based on ISDB-T Standard and using GNU radio for it, I building a fullseg transmiter .
The problem is when trying to run my project, using the blocks of the ISDB-T Module I found in this page, the following massage appears: :""ModuleNotFoundError: No module named 'isdbt'. "" Then I try to use the the following troubleshooting: 

_" Q: I got the following error: "ModuleNotFoundError: No module named 'isdbt'". What's wrong?

A: You probably didn't setup the PYTHONPATH correctly. It should include at least /usr/local/lib/python3/dist-packages. For a system-wide solution, you may edit /etc/environment and include the following line PYTHONPATH="$PYTHONPATH:/usr/local/lib/python3/dist-packages". "_

but it didn't work as well. How should i proceed in order to solve this Issue. Please, I need your answer ASAP!

Best regards!!!

[image]

git-artes, 2022-06-28:
What do you get when you write 
 echo $PYTHONPATH
on the terminal?

Samircaiala, 2022-06-28:
Actually when i right this code "echo $PYTHONPATH" nothing happens, the terminal only opens a new empty line.

git-artes, 2022-06-28:
Then you haven't correctly set the PYTHONPATH variable. Are you sure you've included it in /etc/environment? You may try writing 
PYTHONPATH="$PYTHONPATH:/usr/local/lib/python3/dist-packages"
on a terminal, and then, from that same terminal, opening gnuradio-companion. That way you're sure it's set, but you should try to find a more "permanent" solution.

Samircaiala, 2022-07-01:
Hello! I tried to do what you describe above, and i opened some of the examples given in the gr-isdbt folder by the same terminal and then a try to run the simulation and got this message that i printed below. Is there something else that i could be missing?
[image]

 Cause i even wrote the PYTHONPATH="$PYTHONPATH:/usr/local/lib/python3/dist-packages" in the enviroment file (that actually had to force permission to save the new line in this, cause it could be open only for reading. So i used this code "sudo gedit /etc/environment", to get the temporary permition).

Aishy-artzy, 2022-07-20:
I had the same issues. But i noticed that my python version on gnuradio3.8  was python3.6 and not python3. I tried copying the isdbt folder to that root directory and that removed the module not found error.