git-artes/gr-isdbt #54: Trouble with finding the correct parameters (HackRFOne)

AudioStone, 2022-11-03:
Hi, I've been reviewing your work on the ISDB-TB receiver and I'm having some problems playing the .ts file with ffplay. This is because I can't set the parameters right, to display a constellation similar to the one in the paper (I use a HackRfOne). Can you give me any suggestions please? I'm in Salta - Argentina.
I tried every mode and guard interval and i'm uneable to find a clear peak. 
[image]
[image]

git-artes, 2022-11-03:
hi, 
It seems like the mode is indeed 3, but you're not getting the guard interval right (the plateau in the autocorrelation graph is very big). 
On the other hand, what spectrum are you getting? Is the TV signal strong enough? I see you haven't changed the gain settings.

AudioStone, 2022-11-03:
With the others guard intervals, it just get worse: 
[image]
[image]
I'm getting this spectrum: 

[image]
Should I modify the gain?

git-artes, 2022-11-03:
That's clearly too little power you're getting from the signal. Try increasing the gains or getting closer to the antenna. Are you sure that's the correct frequency?

AudioStone, 2022-11-03:
I´m not actually sure, i was trying with the following list of channels from Argentina: 
Canal 14: 473.143 MHz.
Canal 15: 479.143 MHz.
Canal 16: 485.143 MHz.
Canal 17: 491.143 MHz.
Canal 18: 497.143 MHz.
Canal 19: 503.143 MHz.
Canal 20: 509.143 MHz.
Canal 21: 515.143 MHz.
Canal 22: 521.143 MHz.
Canal 23: 527.143 MHz.
Canal 24: 533.143 MHz.
Canal 25: 539.143 MHz.
Canal 26: 545.143 MHz.
Canal 27: 551.143 MHz.
Canal 28: 557.143 MHz.
Canal 29: 563.143 MHz.
Canal 30: 569.143 MHz.
Canal 31: 575.143 MHz.
Canal 32: 581.143 MHz.
Canal 33: 587.143 MHz.
Canal 34: 593.143 MHz.
Canal 35: 599.143 MHz.
Canal 36: 605.143 MHz.
I will try increasing the gains, but how can I find the correct frequency faster? Is there a way to make a frequency sweep or something like that?

git-artes, 2022-11-04:
You should try with gqrx or any other SDR software that let's you check the spectrum, and try to find a strong signal.

closed 2023-07-14 by git-artes