git-artes/gr-isdbt #55: AttributeError: 'gnuradio.gr.gr_python.logger' object has no attribute 'warning'

hlechner, 2023-06-02:
Following the current system wide installation instructions:
```
git clone https://github.com/git-artes/gr-isdbt.git  
cd gr-isdbt  
mkdir build  
cd build  
cmake ../  
make && sudo make install  
```

no compile error.

I had no major problems, just had to move few files that were installed in:
``/usr/local/share/gnuradio/grc/blocks/`` to ``/usr/share/gnuradio/grc/blocks/``
``/usr/local/lib/`` to ``/usr/lib/``
etc..

However trying to use the ``rx_demo`` I'm having this issue:
```
Generating: '/home/hlechner/Downloads/gr-isdbt/gr-isdbt/examples/rx_demo.py'
>>> Warning: The block 'blocks_throttle_0' is deprecated.

Executing: /usr/bin/python3 -u /home/hlechner/Downloads/gr-isdbt/gr-isdbt/examples/rx_demo.py

Traceback (most recent call last):
  File "/home/hlechner/Downloads/gr-isdbt/gr-isdbt/examples/rx_demo.py", line 344, in <module>
    main()
  File "/home/hlechner/Downloads/gr-isdbt/gr-isdbt/examples/rx_demo.py", line 315, in main
    gr.logger("realtime").warning("Error: failed to enable real-time scheduling.")
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'gnuradio.gr.gr_python.logger' object has no attribute 'warning'

>>> Done (return code 1)
```

Not sure if it's related but I found this:
https://wiki.gnuradio.org/index.php/Logging
``In GNU Radio 3.10, you construct your own logger from gr.logger.``

I'm using the GNU Radio 3.10.6 (from arch linux repository)

m-a-aniskovich, 2023-07-11:
Looks like ```gr.logger("realtime").warning()``` does not work anymore in GNU Radio 3.10+. It was changed to ```gr.logger("realtime").warn()```. But developers did not change this behavior in GNURadio Companion compiler. So after compiling .grc to .py file i change all occurencies of ```.warning(``` to ```.warn(``` by hand.

git-artes, 2023-07-14:
Sorry for not answering, but I've been quiet busy and couldn't test the problem. Thanks to @m-a-aniskovich for providing a solution. If I understood correctly, it should be solved in the newer versions of GNU Radio.

hlechner, 2023-08-24:
Since it's a GNU Radio problem, should I close the issue?

git-artes, 2023-08-24:
Will do. Thanks!

closed 2023-08-24 by git-artes