git-artes/gr-isdbt #57: 1seg tx example

VasylSamoilov, 2023-09-22:
1seg example is missing, and it's very hard to find proper stream parameters for 1seg TS. I'm trying to get it to work with nintendo ds 1seg tuner and psp 1seg tuner, but no luck. Can soneone point me in the correct documentation (and maybe someone has TS stream capture with 1seg to test with? That would be beneficial).
Thank you.

git-artes, 2023-09-29:
Hi. There were a few attempts in the past to make it work with Nintendo DS or PSP (i.e. https://www.journaldulapin.com/2021/09/30/1seg-eyetv/). Not sure if it worked. If you want a 1-seg TS, you may decode the example we've included, which includes a 1-seg transmission. That should work.

VasylSamoilov, 2023-09-30:
Hi. Thank you, it worked, but suffers from shuttering.
I used this TS as a reference
https://tsduck.io/streams/japan-dttv/AAB.ts

For test.ts I filtered out (with tsduck) pids relevant to 1seg service id,
for test2.ts I added "Global PID's" to filter - this added EPG and
presumably subtitles, clock setting and maybe something else.

The resulting shuttering seems to be related to stream or to transmission
settings. I tested on bare metal, in real time and with file as output -
and shuttering is very consistent.
Can you please have a look, maybe you can point me in the right direction
which transmission or stream settings should be adjusted?

Relevant files are uploaded to google drive:
https://drive.google.com/drive/folders/1DeIa3urPgdAIEHY6kUvwIR90chNYUdIo?usp=sharing

Thank you.

git-artes, 2023-10-06:
Why are you feeding only layer A to the transmitter? Not sure what it will do in that case...

VasylSamoilov, 2023-10-07:
Well, I am trying (and kinda got it mostly operational) to receive 1seg portable device transmission on actual hardware, in my case it's Nintendo DS and PSP. These devices receive only 1seg, QPSK, japanese ISDB-t. According to documents I found layer A is 1-seg, QPSK, and layer B is 12-seg 64QAM, so I'm transmitting only what portable devices are able to receive, which is 1-seg layer A. Problem is that reception is suffering from some overruns or underruns and both devices are dropping video and audio streams. Question is how Japanese 1seg is different and what parameters can be tunes so 1seg reception will be more stable. Given that original transport stream is full 13-seg from actual transmission, maybe I incorrectly repackaged the stream. Or maybe limesdr is the problem (wrong bandwidth or sampling rate). Any help would be appreciated.

VasylSamoilov, 2023-10-25:
Well, after some trial and error I came to conclusion that captured transport stream is (obviously) layers A and B combined, but it's beyond my skills to properly filter out and separate this layers. Whatever PIDs I "filter" result in TS that will desync on real hardware due to improper timings of special packets. Generating TS that will be received on real japanese 1seg hardware is also tricky, because it requires some channel/program metadata in japanese ISDB ARIB-STD-B24 character encoding and in specific format, which I can't properly encode with ffmpeg.

So, if anyone will be willing to give it a try, there are two different problems to solve:
1) Given a valid TS captured from real broadcast (example: https://tsduck.io/streams/japan-dttv/AAB.ts) - how to split it into proper layer A and layer B ts files to re-broadcast properly with example files provided so they will be timed and interleaved correctly.
2) How to generate proper layer A and B TS files (codecs, bit rate, etc) and what will be the minimal metadata required to recieve japanese ISDB with ARIB-STD-B24 character encoding on real hardware.