git-artes/gr-isdbt #58: Error Cmake

ignacioab98, 2023-11-07:
Al hacer cmake en cd build me tira el siguiente error:

**CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:221 (message):
  Python bindings for bit_deinterleaver.h are out of sync
Call Stack (most recent call first):
  python/isdbt/bindings/CMakeLists.txt:53 ### (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!**

Mi version de gnuradio es 3.10.8.0 (Python 3.11.6). 

Cualquier ayuda es agradecida.

SebasGv150402, 2023-11-16:
https://github.com/git-artes/gr-tempest/issues/24
me funciono añadiendo uno por uno, este link te ayudara. Consejo, usa el comando en la parte de cd gr-isdbt y el comando ponlo asi: gr_modtool ....

mdoallo, 2024-07-11:
Prueba de ejecutar en /gr-isdbt/ estos comandos:

```
gr_modtool bind -u bit_deinterleaver
gr_modtool bind -u byte_deinterleaver
gr_modtool bind -u byte_interleaver
gr_modtool bind -u carrier_modulation
gr_modtool bind -u energy_descrambler
gr_modtool bind -u energy_dispersal
gr_modtool bind -u frequency_deinterleaver
gr_modtool bind -u frequency_interleaver
gr_modtool bind -u hierarchical_combinator
gr_modtool bind -u ofdm_synchronization
gr_modtool bind -u pilot_signals
gr_modtool bind -u reed_solomon_dec_isdbt
gr_modtool bind -u subset_of_carriers
gr_modtool bind -u symbol_demapper
gr_modtool bind -u time_deinterleaver
gr_modtool bind -u time_interleaver
gr_modtool bind -u tmcc_decoder
gr_modtool bind -u tmcc_encoder
gr_modtool bind -u viterbi_decoder
```

Luego cambia a /build/ para continuar con el 
`cmake ../`

git-artes, 2025-01-30:
Perdón la demora (más de un año!!!). Esto debería haber quedado resuelto en la versión nueva que está ahora en el repo.

closed 2025-01-30 by git-artes