git-artes/gr-isdbt #59: python bindings for bit_deinterleaver.h are out of sync

SebasGv150402, 2023-11-14:
Hi, I am trying to install the isdbt's blocks for gnu radio 3.10 (I tried to install 3.8 but no success) and when I run the cmake ../ command I get the following error, I tried many methods but nothing was fixed. Any help please? (Im using Virtual Box with Ubuntu 22.04)

cmake error at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/grpybind.cmake:221 (message):
python bindings for bit_deinterleaver.h are out of sync
call stack (most recent call first):
python/isdbt/bindings/cmakelists.txt:53 (gr_pybind_make_oot)

git-artes, 2023-11-14:
Hi. What problem did you have with the 3.8 version? That should work perfectly.

SebasGv150402, 2023-11-15:
Hello! Well, I looked for how to install 3.8 and followed the steps to install it, but every time I did it the version that was installed was 3.10, any recommendation? maybe another way to install or something? If i install a oldest version of ubuntu, could i install 3.8?

Paulo-D2000, 2024-01-24:
Hi, sorry for this... please check https://github.com/git-artes/gr-isdbt/pull/60/commits/49e9caf85964a060d98aadb01ee848c60b394720

git-artes, 2025-01-30:
@Paulo-D2000 is right, and his idea is now included in the repo. Thanks!

closed 2025-01-30 by git-artes