git-artes/gr-isdbt #64: Support for new tuner device and ffmpeg handling

maurirope, 2024-11-13:
Hello, so i managed to run your software a little, i do get like some static playing so I guess it is on the right track however i have yet to understand how to tweak your setup so that i could try and use the device i do have.

I am currently running a Ubuntu 24.04.1 and I have purchased here in Uruguay, LEDSTAR ISDB-T tuner USB bla bla. Ubuntu is running as VM within Proxmox node. I pass through USB device to the VM and can lsusb and see device 1df7:2500. I kinda followed [the instructions here](https://revspace.nl/Msi2500SDR) to get [the soapy driver installed](https://github.com/ericek111/SoapyMiri) and i can "probe" it in gnuradio. Then I managed to run your example RX receiver.

I have soapy SDR source

[image]
[image]
[image]
[image]

Now i cant just run it from gnuradio because there are some things that it ends up putting into the resulting script that are not quite compatible:

Line 120
`self.soapy_sdrplay_source_0.write_setting('agc_setpoint', 0)`
Returns invalid setting agc_setpoint even though soapy probe says AGC supported true and is set true in source dialog

When commented script continues fine and then we hit some QThread error, fixed running script as sudo

Line 107
`self.set_soapy_sdrplay_source_0.set_gain(channel, 'RFGR', min(max(lna_state, 0), 9))`
Changing RFGR to LNA allows continuing script execution.

then Line 257, is problem with non existing directory to write the .ts file because of location of my saved script, changed location to ./file

now Lo and behold!!!!

[image]

We going somewhere!!!! No idea where, but there is some output...

Then I tried to pipe that thing into ffmpeg

`sudo python3 -u script.py | ffplay -f f32le -ar 48000 -acodec pcm_f32le -`

[image]

 but i have no idea what im doing at this point....

Now my resulting script is attached for your analysis, Im sure im getting the sizing of the blocks and bits and whatnot all wrong. also im probably needing to strip somethings out of the output before passing it to ffmpeg, but it has exceeeded my knowledges, maybe you guys can help out a bit

[rx_demo2.txt](https://github.com/user-attachments/files/17724362/rx_demo2.txt)

I also attach sopy miri driver probe of the dongle

[Soapy SDR the SDR abstraction.txt](https://github.com/user-attachments/files/17724365/Soapy.SDR.the.SDR.abstraction.txt)

git-artes, 2024-11-13:
Hi,
I can't help you with soapy, but I can help you with gr-isdbt. You're getting several characters on the terminal, which probably means you're dropping samples (you should check the documentation of the hardware you're using). That's not good. You should try to give more resources to the VM (or run gr-isdbt natively). 
The other thing is the ffmpeg command you're using. You should use  ffplay on whatever file's on the the end of the flowgraph (and whatever layer you're using).
However, some checking you should do first: 
 - Did you get a nice spectrum? I'm seeing a 0.2 MHz bandwidth in the parameters of soapy... It should be at least 2.9 MHz, as in the LPF in the flowgraph.
 -  Are you getting a constellation? There is a specific flowgraph for checking that out.
 - Are you using the correct parameters for the channel? Since you're here in Uruguay, it should work out of the box, but you should run the getting_the_parameters script both to check that the signal is reasonable and that nothing has changed since last time I've run it.

best of lucks!

maurirope, 2024-11-13:
Thank you very much! These are very nice pointers to get started into researching further! Will come back with knowledges to share if reach any conclusion! Have a very nice day

maurirope, 2024-11-13:
Oh I remembered I should add, im not directly piping the console output to ffmpeg but rather outputting the byte stream that shouldve been written to the ts file to the output instead, and that is what is being piped through bash to ffplay

closed 2024-11-13 by maurirope

maurirope, 2024-11-14:
Also it has 8 cores assigned, 16gb ram, q35 guest tools, drivers are virtio, running on xeon 2697v3 with native host instructions exposed