git-artes/gr-isdbt #66: Example for 1seg in master is missing

justalemon, 2025-02-27:
Hi! I tried to run this in Arch Linux with almost success. Everything compiles and installs just fine.

I tried to run the 1seg example but there is none in master, so I tried to use the [3.7 1seg example](https://github.com/git-artes/gr-isdbt/blob/maint-3.7/examples/rx_1seg_demo.grc) on 3.10 but some parts were missing:

[image]

Would it be possible to update the 1seg example for compatibility with the master branch? Thanks!

PS: I barely know the basics so this is out of my area of expertise.

git-artes, 2025-03-24:
That's a great idea, but I'd have to find time to do it. It requires more than simply copy-pasting code since migrating from 3.7 to 3.8+ has several changes. But I'll try to find time since it's true that it's usable with an rtl-sdr.

dragoon2014, 2026-02-15:
Hi, I submitted PR.
Please check it in your free time!
https://github.com/git-artes/gr-isdbt/pull/70