git-artes/gr-isdbt #67: constellation is noisy. obtaining_parameters shows TMCC OK. but rx_demo does not produce a viable .ts

GastonMazzei, 2025-05-16:
Hello, 

Thank you for your effort, great project. 

I wanted to share my experience to understand if the usage is correct and/or if there is a bug. 

I'm using a file source which I COULD upload to GDrive if requested. 
Its a float capture with an Ettus B200 mini using the following command: 

`rx_samples_to_file  --channels="0" --type float --rate 8126984  --freq 605143000  --bw 6000000 --gain 44  --nsamps=8126984    capture.iq`

The command above produces a capture that is 1 second long. 
I tested longer captures, albeit we have a transfer rate where we see some overflow after more than e.g. 10 seconds of continuous capture. 
The B200-mini is a Chinese clone, and transports data over USB3.0.
We tried both using the USB 3.0 port of our motherboard, and adding a dedicated PCIe 8x to USB 3.0 converted on a dedicated PCIe lane... but this is not the point of the issue. 

Our antenna used is an omnidirectional helical copper coil antenna, similar to the one on the pic attached. 
Gain could be 5dB, and we are indoors close to some metal structures... less than ideal, I know.
[image]

The only thing changed on viewing_the_constellation is the guad interval set to 1/8. Mode is MODE3 8K, central freq is 605.143 MHz which here in Argentina (Buenos Aires) is 36.1 "America TV". I think there are different towers broadcasting it at the same time, aprox 20 km apart. 
Its one of the strongest channels here... 

Please see results and warnings below
Executing: /usr/bin/python3 -u /home/gr-isdbt/examples/viewing_the_constellation.py
realtime :warning: Error: failed to enable real-time scheduling.
buffer_double_mapped :warning: allocate_buffer: tried to allocate   4 items of size 44936. Due to alignment requirements   512 were allocated.  If this isn't OK, consider padding   your structure to a power-of-two bytes.   On this platform, our allocation granularity is 4096 bytes.
buffer_double_mapped :warning: allocate_buffer: tried to allocate   18 items of size 44936. Due to alignment requirements   512 were allocated.  If this isn't OK, consider padding   your structure to a power-of-two bytes.   On this platform, our allocation granularity is 4096 bytes.
buffer_double_mapped :warning: allocate_buffer: tried to allocate   4 items of size 44936. Due to alignment requirements   512 were allocated.  If this isn't OK, consider padding   your structure to a power-of-two bytes.   On this platform, our allocation granularity is 4096 bytes.
buffer_double_mapped :warning: allocate_buffer: tried to allocate   16 items of size 22468. Due to alignment requirements   1024 were allocated.  If this isn't OK, consider padding   your structure to a power-of-two bytes.   On this platform, our allocation granularity is 4096 bytes.

[image]

[image]

[image]

and also please check this constellation in a case where we decreased 20dB the amplification in order to confirm the noise was not added by us (we are not checking if there is saturation, sorry for being lazy, but we explored all the possible gain ranges and haven't found a clean constellation in any region)
[image]

Regarding obtaining_parameters:
we changed the OFDM decoding to be 1/8 and 8K (by default it was 1/4 I believe). 
The correlation confirmed this was the best setup :-) you can look at it below... anyways... we know this is the correct setup here in BsAs, Arg 

The TMCC output is as follows:

							TMCC ANALYSIS						
---------------------------------------------------------------------------------------------------------
Layer								: A
Carrier Modulation Scheme			: QPSK
Convolutional Code Rate			: 2/3
Time Interleaving Length			: 8(mode 1), 4(mode 2), 2(mode 3)
Number of segments for this layer	: 1
--------------------------------------------------------------------------------------------------------
Layer								: B
Carrier Modulation Scheme			: 64QAM
Convolutional Code Rate			: 2/3
Time Interleaving Length			: 16(mode 1), 8(mode 2), 4(mode 3)
Number of segments for this layer	: 12
--------------------------------------------------------------------------------------------------------
Layer								: C
Carrier Modulation Scheme			: UNUSED
Convolutional Code Rate			: UNUSED
Time Interleaving Length			: UNUSED
Number of segments for this layer	: UNUSED
--------------------------------------------------------------------------------------------------------
TMCC OK

							TMCC ANALYSIS						
---------------------------------------------------------------------------------------------------------
[... 138 lines cut ...]
Number of segments for this layer	: 1
--------------------------------------------------------------------------------------------------------
Layer								: B
Carrier Modulation Scheme			: 64QAM
Convolutional Code Rate			: 2/3
^CTime Interleaving Length			: 16(mode 1), 8(mode 2), 4(mode 3)
Number of segments for this layer	: 12
--------------------------------------------------------------------------------------------------------
Layer								: C
Carrier Modulation Scheme			: UNUSED
Convolutional Code Rate			: UNUSED
Time Interleaving Length			: UNUSED
Number of segments for this layer	: UNUSED
--------------------------------------------------------------------------------------------------------
TMCC OK

[image]
[image]
[image]

Finally let me show the output of rx_demo, as I don't know what values mean here 

[image]

We changed the file sink output path which I guess it's completely transparent to the results... and it is a file that is either of size 0 or very small (e.g. 4000 bytes), and playing it with "vlc" from command line gives a zero-length video.

GastonMazzei, 2025-05-16:
Update: in our case its better to have the omni antenna outdoors with a 3m cable run, as opposed to indoors with a 50cm cable run. Look at the constellation in the case of 3m outdoors (I dont know which type of cable is, but I think its 75ohm matched even tho the SDR is 50ohm)

[image]

git-artes, 2025-05-16:
Hi, great to hear this good news. Quick question: are you configuring all other blocks are indicated by the TMCC?

GastonMazzei, 2025-05-16:
You are right, it was misconfigured. 

This is the TMCC decoding of the last (best) constellation. 
Its only one "case" (its called frame?) but they are all the same I guess 
```
							TMCC ANALYSIS						
---------------------------------------------------------------------------------------------------------
Layer								: A
Carrier Modulation Scheme			: QPSK
Convolutional Code Rate			: 2/3
Time Interleaving Length			: 8(mode 1), 4(mode 2), 2(mode 3)
Number of segments for this layer	: 1
--------------------------------------------------------------------------------------------------------
Layer								: B
Carrier Modulation Scheme			: 64QAM
Convolutional Code Rate			: 2/3
Time Interleaving Length			: 16(mode 1), 8(mode 2), 4(mode 3)
Number of segments for this layer	: 12
--------------------------------------------------------------------------------------------------------
Layer								: C
Carrier Modulation Scheme			: UNUSED
Convolutional Code Rate			: UNUSED
Time Interleaving Length			: UNUSED
Number of segments for this layer	: UNUSED
```

This required to change the Time Deinterleaver block 😄 

The RF sample is 1 second long. 
File sink is unbuffered: off (i.e. buffered)
After 1 minute the rx_demo is still running 
results_layer_b.ts is already 60Mb in size. Im stopping it. 
Vlc player complains:

I guess we should NOT run this on loop but a single iteration? 

```

vlc persistent/result_layer_b.ts 
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[0000637537e07550] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0
[... cut ...]
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 6, expected 5) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 7) for PID 257
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 3, expected 6) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 5, expected 4) for PID 259
[000071717c0011a0] ts demux error: libdvbpsi error (PSI decoder): TS duplicate (received 8, expected 9) for PID 0

```

GastonMazzei, 2025-05-16:
Update: 

So I removed the loop (repeat: false) in the file source and the execution runs for a fixed (short) amount of time as expected. 

The produced layer B ts file is 250kB long for a 1s radio capture, but vlc media when trying to play it complains:
```

vlc persistent/result_layer_b.ts 
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[00005f3fd83dc550] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007a38c8c19d40] main decoder error: buffer deadlock prevented
[00007a38c91c6150] main decoder error: buffer deadlock prevented
[00007a38fcc0ff80] main decoder error: buffer deadlock prevented
[00007a38fd1be8c0] main decoder error: buffer deadlock prevented
[00007a3908c19b80] main decoder error: buffer deadlock prevented
[00007a39091aea20] main decoder error: buffer deadlock prevented
[00007a3908c0a1c0] main decoder error: buffer deadlock prevented
[00007a39091aed60] main decoder error: buffer deadlock prevented
[00007a3908c0dfb0] main decoder error: buffer deadlock prevented
[00007a39091aed80] main decoder error: buffer deadlock prevented
[00007a3908c0bc90] main decoder error: buffer deadlock prevented
[00007a39090daa40] main decoder error: buffer deadlock prevented
```

git-artes, 2025-05-16:
Great. Yes, they are all the same, except naturally when the station changes the channel's configuration. You may safely change the "print params" to False in the TMCC decoder now. 

What pre and post-Viterbi BER are you getting now? In any case, you should use ffmpeg to play the resulting transport stream (after installing ffmpeg, the command is ffplay)

GastonMazzei, 2025-05-16:
Hi, 

post-Viterbi BER (log 10) is very constan around -0.3
Is that OK? 

ffplay shows this output 
```

ffplay persistent/result_layer_b.ts 
ffplay version 6.1.1-3ubuntu5 Copyright (c) 2003-2023 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.2.0-23ubuntu3)
  configuration: --prefix=/usr --extra-version=3ubuntu5 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --disable-omx --enable-gnutls --enable-libaom --enable-libass --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-openal --enable-opencl --enable-opengl --disable-sndio --enable-libvpl --disable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-ladspa --enable-libbluray --enable-libjack --enable-libpulse --enable-librabbitmq --enable-librist --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libx264 --enable-libzmq --enable-libzvbi --enable-lv2 --enable-sdl2 --enable-libplacebo --enable-librav1e --enable-pocketsphinx --enable-librsvg --enable-libjxl --enable-shared
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[mpegts @ 0x704170000c80] PES packet size mismatchsq=    0B f=0/0   
[mpegts @ 0x704170000c80] Packet corrupt (stream = 3, dts = 3711054464).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810508064).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810513464).
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[mpegts @ 0x704170000c80] PES packet size mismatch
[mpegts @ 0x704170000c80] Packet corrupt (stream = 1, dts = 3810455278).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810515264).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810522464).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810526064).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810531464).
[mpegts @ 0x704170000c80] PES packet size mismatch
[mpegts @ 0x704170000c80] Packet corrupt (stream = 1, dts = 3810478318).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 3, dts = NOPTS).
[mpegts @ 0x704170000c80] PES packet size mismatch
[mpegts @ 0x704170000c80] Packet corrupt (stream = 4, dts = 3711022684).
[mpegts @ 0x704170000c80] probed stream 1 failed
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
[NULL @ 0x7041700f80c0] non-existing PPS 0 referenced
[h264 @ 0x704170008fc0] non-existing PPS 0 referenced
[h264 @ 0x704170008fc0] decode_slice_header error
[h264 @ 0x704170008fc0] no frame!
[h264 @ 0x7041700f80c0] non-existing PPS 0 referenced
[h264 @ 0x7041700f80c0] decode_slice_header error
[h264 @ 0x7041700f80c0] no frame!
[aac @ 0x704170101400] decode_band_types: Input buffer exhausted before END element found
[mpegts @ 0x704170000c80] PES packet size mismatch
[mpegts @ 0x704170000c80] Packet corrupt (stream = 3, dts = 3711054464).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810508064).
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810513464).
[mpegts @ 0x704170000c80] PES packet size mismatch
[... 36 lines cut ...]
[NULL @ 0x704170008fc0] non-existing PPS 0 referenced
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810526064).
[NULL @ 0x704170008fc0] non-existing PPS 0 referenced
[mpegts @ 0x704170000c80] Packet corrupt (stream = 0, dts = 3810531464).
[NULL @ 0x704170008fc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x70417016bb40] non-existing PPS 0 referenced
[h264 @ 0x70417016bb40] decode_slice_header error
[h264 @ 0x70417016bb40] no frame!
[h264 @ 0x704170026300] non-existing PPS 0 referenced
[h264 @ 0x704170026300] decode_slice_header error
[h264 @ 0x704170026300] no frame!
[h264 @ 0x704170168080] non-existing PPS 0 referenced
[h264 @ 0x704170168080] decode_slice_header error
[h264 @ 0x704170168080] no frame!
[h264 @ 0x7041700ea800] non-existing PPS 0 referenced
[h264 @ 0x7041700ea800] decode_slice_header error
[h264 @ 0x7041700ea800] no frame!
[h264 @ 0x7041701cb380] non-existing PPS 0 referenced
[h264 @ 0x7041701cb380] decode_slice_header error
[h264 @ 0x7041701cb380] no frame!
[h264 @ 0x7041701df240] non-existing PPS 0 referenced
[h264 @ 0x7041701df240] decode_slice_header error
[h264 @ 0x7041701df240] no frame!
```

git-artes, 2025-05-16:
-0.3 is too high. You should get something like -4 at least (i.e. a bit error rate of about 10^-4) for the video to be playable. With that level of noise it will be difficult to improve. Did you try decoding the 1-seg only? That should work with the signal you're getting since it uses QPSK.

GastonMazzei, 2025-05-16:
Hi, thanks for the info. And thanks for the help :) 
Indeed, it all points out to my frontend having a really bad signal to noise ratio. 
Im confused. Adding an LNA helped marginally, but still its very noisy: maybe today being a rainy day is not the best day to test this.

Something related to your last comment: the 1-seg I referred to is just an IQ capture I did of 1 second of duration 😄 it also has QAM64 in the Layer B. I guess you refer to decoding layer A which uses QPSK apparently... but how? just plugging the bit interleaver to the layer A instead of B out of the Symbol Demapper and changing parameters accordingly? 
Im a real noob on this... there is digital TV info on both layer A and B, with different qualities? thanks

git-artes, 2025-05-16:
Great. Regarding the setup, for starters, I use a USRP b200-mini, but original, not a Chinese clone. Don't take this as a brag. I've heard those clones are not good at all. Plus, my attempts at using other SDR hardware have not been as satisfactory. 

About the antenna, I've used a very simple dipole inside the university and got a really nice constellation. No LNA needed.

GastonMazzei, 2025-05-17:
Just for curiosity, when you capture (outside Gnuradio, and save to a file) are you setting gain control to automatic?

git-artes, 2025-05-19:
No, I tend to use a fixed gain, typically in between max and min gains (or the highest that does not distort the signal).

GastonMazzei, 2025-05-19:
Hi. 
Thanks.
 
We bought a LPF at 700Mhz to remove LTE noise. It's not here yet. 
We also have another (Chinese copy of a) B200 that exposes REF_CLK 10Mhz that we will try asap using our GPSDO 10Mhz 3.3Vpp (So far tests were without external REF CLK).

I also tested examples from the gnuradio wifi project and the situation is similar:
- noisy constellation 
- decoding succeeds for some messages. Mostly control channel.

Summary:
The current hypotheses as why this cannot fully work with our hardware/software is not well-formed yet, but it's something we are paying attention to and should be resolved before the end of the month. 

If you think about something we might be missing it would be much appreciated.
I will keep this thread updated. Thanks.

git-artes, 2025-06-02:
Hi,
The problem in the first image is that you're dropping a lot of samples (the "O"s in the terminal). Plus, the spectrum doesn't look so nice either. I don't think it has nothing to do with how many TSs are in that particular channel. We haven't tested with so many TSs, but given that that's an "upper layer" problem, I don't think it will impact the performance of the OFDM receiver.

good luck!